OpenXiangShan/XiangShan #3942: Incorrect Behavior of `hvip` Bits 13 to 15

youzi27, 2024-11-26:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

Hi there,

The RISC-V specification states that bits 11 to 15 of the `hvip` register should be zero. However, it has been observed that on the XiangShan, bits 13 to 15 of `hvip` can be written with 1. This behavior also exists in NEMU.

In the pre-refactoring CSR code, it appears that `hvipMask` was used to handle these fields, ensuring that the default read-only zero behavior was enforced. 
https://github.com/OpenXiangShan/XiangShan/blob/3956160f98fd99a914dc3ed63049918802875cfa/src/main/scala/xiangshan/backend/fu/CSR.scala#L593

Currently, this masking or validation logic seems to be missing, allowing writes to these reserved fields.

If this is an intentional design choice for XiangShan, please let me know. Any clarification on this behavior would be greatly appreciated.

**Screenshots**
[image]

### Expected behavior

Bits 11 to 15 of the `hvip` register should be read-only zero.

[image]

### To Reproduce

In S-mode, attempt to write all 1 to the `hvip` register.

### Environment

- ready-to-run's Spike commit id: [OpenXiangShan/ready-ro-run@3575e65](https://github.com/OpenXiangShan/ready-to-run/commit/3575e65) (Date: Fri Nov 22 17:06:53 2024 +0800) 
- XiangShan commit id: https://github.com/OpenXiangShan/XiangShan/commit/aecf601e803bfd2371667a3fb60bfcd83c333027 (Date: Tue Nov 19 16:16:35 2024 +0800)

NewPaulWalker, 2024-11-26:
Thank you for your issue! Would you mind sharing your test case?

youzi27, 2024-11-26:
Hi @NewPaulWalker ,

Thank you for your response. The test case to reproduce the issue is: [testcase.zip](https://github.com/user-attachments/files/17918813/testcase.zip)

NewPaulWalker, 2024-11-27:
> Currently, this masking or validation logic seems to be missing, allowing writes to these reserved fields.
> 
> If this is an intentional design choice for XiangShan, please let me know. Any clarification on this behavior would be greatly appreciated.

Thank you for sharing your test. This behavior is actually expected in XiangShan. Since XiangShan implements the **AIA** (RISC-V Advanced Interrupt Architecture), bits 13–64 of **hvip** are writable.
https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/backend/fu/NewCSR/HypervisorLevel.scala#L224

Additionally, when the **Sscofpmf** extension is implemented, the **LCOFI** bit (bit 13) of **hvip** becomes implemented and writable.  
[image]

However, I noticed that in Spike, this bit of **hvip** does not seem to be considered. I have submitted a PR to upstream to see if this issue can be addressed: https://github.com/riscv-software-src/riscv-isa-sim/pull/1862

Moreover, there appears to be a bug in NEMU: the **LCOFI** bit of **hvip** depends on **Shlcofideleg** instead of **Sscofpmf**. I have already made the necessary corrections: https://github.com/OpenXiangShan/NEMU/pull/687

youzi27, 2024-11-27:
😊Thank you for your patient explanation and the PRs submitted for NEMU and Spike.

NewPaulWalker, 2024-12-06:
Please use this commit https://github.com/OpenXiangShan/XiangShan/commit/a60d667e2e8568a26fbfec72b3e9c484df525995 to test again.

youzi27, 2024-12-09:
The test passed, so I’m closing this issue. Thanks!

closed 2024-12-09 by youzi27