OpenXiangShan/XiangShan #3943: Unable to Set `stvec` Mode to 1 (Vectored)

youzi27, 2024-11-26:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hi there,

I would like to confirm whether XiangShan supports setting the `stvec` register's mode to 1 (Vectored). Based on my review of the source code, it seems that this mode is supported. However, I am unable to write 1 to the mode field of `stvec.`

https://github.com/OpenXiangShan/XiangShan/blob/3956160f98fd99a914dc3ed63049918802875cfa/src/main/scala/xiangshan/backend/fu/NewCSR/CSRDefines.scala#L153-L158

Could you please confirm whether this is intentional or if additional steps are required to enable the Vectored mode?

Thank you for your assistance! Let me know if further details or examples are needed.

youzi27, 2024-11-26:
Hi @NewPaulWalker ,

Thank you for your response. When I attempted to read and write the `mode` field of `xstvec` using the `csrrsi` instruction, the mismatch case occurred with the instruction `csrrsi s2, xstvec, 31`. The meaning of this instruction is to perform a bitwise OR between the CSR and the immediate value (uimm) and write the result back to the CSR. 

Is this implementation intentional in XiangShan? Intuitively, when the CSR and uimm are bitwise OR-ed in the `csrrsi` instruction, the reserved bits should be automatically masked.

NewPaulWalker, 2024-11-27:
> Thank you for your response. When I attempted to read and write the `mode` field of `xstvec` using the `csrrsi` instruction, the mismatch case occurred with the instruction `csrrsi s2, xstvec, 31`. [...]

Thank you for sharing your testing method. In fact, you are attempting to write the value 3 to the MODE field of **xtvec**. However, **mtvec** is a WARL (Write-Any Read-Legal) CSR, and its MODE field reserves values greater than or equal to 2. 
[image]

Therefore, your attempt to write 3 to the MODE field will be ignored. This behavior is intentional in XiangShan, and NEMU is designed this way as well.
https://github.com/OpenXiangShan/NEMU/blob/master/src/isa/riscv64/system/priv.c#L828
Additionally, for **mtvec**, there are no reserved bits; instead, it has reserved values for the MODE field.

youzi27, 2024-11-27:
Thank you for your patient explanation regarding this question. I will close it now. Thanks!

closed 2024-11-27 by youzi27