OpenXiangShan/XiangShan #3957: Uninitialized `htimedelta` CSR

youzi27, 2024-11-28:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual). 我已经阅读过 RISC-V 指令集手册。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the feature you'd like

I noticed that in XiangShan, the `htimedelta` register isn’t explicitly initialized and instead gets assigned a random value during the build process. Since this register represents the difference between the `time` CSR and the values in VS-mode or VU-mode, I’m wondering if relying on a random value is truly reliable.

Even putting that aside, both NEMU and SPIKE initialize `htimedelta` to 0. To stay aligned with them and avoid potential issues down the road, it might be worth considering fixing this.

Thanks for your time!

NewPaulWalker, 2024-11-29:
Thank you for your issue. `htimedelta` is not a CSR that needs to be explicitly initialized by hardware. Although its initial value is 0 in NEMU, this is not because we explicitly initialize it to 0. Instead, this CSR is a global variable in the NEMU C program, and it gets automatically initialized to 0 during the program initialization.

You can refer to this https://github.com/OpenXiangShan/XiangShan/issues/3882

youzi27, 2024-11-29:
Got it, thanks @NewPaulWalker !

closed 2024-11-29 by youzi27