OpenXiangShan/XiangShan #3959: Unable to Handle Specific Sequences of Illegal Instructions

youzi27, 2024-11-29:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

Hi there,

I have discovered that in M-mode, XiangShan fails to handle **specific sequences** of illegal instructions correctly. When an illegal instruction exception is triggered by such a sequence, the XS does not function as expected, and execution is disrupted.

Although the illegal instruction involves accessing a custom register, a review of the code shows that XiangShan does not define this custom register. Additionally, logs indicate that XiangShan does raise an illegal instruction exception but fails to successfully enter the exception handler.

**Screenshots**
When spike as a ref:
[image]

When nemu as a ref:
[image]

### Expected behavior

The XS should handle exceptions triggered by specific sequences of illegal instructions and continue execution, consistent with the behavior of NEMU and SPIKE.

### To Reproduce

testcase: [ill-test.zip](https://github.com/user-attachments/files/17954627/ill-test.zip)

### Environment

- XiangShan commit id: edb1dfa (Date:   Wed Nov 27 13:40:59 2024 +0800)
- Ready-to-run (NEMU & SPIKE) commit id: https://github.com/OpenXiangShan/ready-to-run/commit/c1dc496 (Date:   Thu Nov 28 12:58:07 2024 +0800)

my-mayfly, 2024-11-29:
This issue is currently being analyzed and processed.

ngc7331, 2024-11-29:
It doesn't look like a Frontend bug, after executing the instruction at `0x80001040`, the configuration of the PMA changed (`addr=0x00000E008000->0x0013341026F3`). Then, the subsequent fetch request hit the PMA entry, which didn't have executable permissions(cfg_x), and then triggered the access fault. I'm not very familiar with PMA, can anyone help?

[image]

ngc7331, 2024-11-29:
So far as I see, 0x7cf is actually the address of the 7th machine-level Pmaaddr register:
https://github.com/OpenXiangShan/XiangShan/blob/7071df62b8073cb0ad5b948b6762ec099408c3d2/src/main/scala/xiangshan/backend/fu/util/CSRConst.scala#L184-L186

And 0x80001040 is `csrrs s4, 0x7cf, t0`, so I guess the behavior is expected?

youzi27, 2024-11-29:
Hi @ngc7331 ,

Thank you for your patient analysis. I agree with your point that this shouldn't be considered a frontend bug, but I still believe it is a bug.

After removing unrelated instructions (e.g., `sraw t3, a2, a2` after the main function), the decoding result of the exception handler differs(not `0x00000000`).

Tang-Haojin, 2024-12-01:
Should be resolved by #3966. Please have a try.

closed 2024-12-02 by Tang-Haojin