OpenXiangShan/XiangShan #4017: Store-load violation checking for vector instruction

jlong299, 2024-12-10:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hi, I am confused about how XS implement store-load violation checking for vector instructions.
In my understanding, one entry of load queue contains one vector micro-op, which may include multiple memory addresses under strided or indexed load scheme. How does the store request compare with all these discrete addresses for store-load violation checking?  So far, I haven't found any code about that comparisons. 
Thank you.

Anzooooo, 2024-12-11:
For vector instructions, we split into multiple `Uop` (micro-operations), and we choose to split complex vector operations into access by element, for example, a vector Idx instruction may generate 4 Uop, each Uop is split into 8 access , for a total of 32 access.
For complex vector instructions, each element occupies a load queue, which guarantees violation detection, i.e., for an Idx instruction, it may occupy 32 or 64 LoadQueues.
For simple and fast vector instructions, such as `Unit-Stride`, one Uop occupies at most two Load Queues, depending on whether the bandwidth of the Load is crossed or not.

jlong299, 2024-12-11:
Got it. Thank you so much!

closed 2024-12-11 by jlong299