OpenXiangShan/XiangShan #4021: worload problem

quanchenliu, 2024-12-10:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

I executed the command `./build/emu -i $NOOP_HOME/ready-to-run/linux.bin` and got the following output, but **I did not see the  OpenSBI LOGO**. The output of the command is provided below.

> $ ./build/emu -i $NOOP_HOME/ready-to-run/linux.bin
> emu compiled at Dec 10 2024, 19:42:48
> Using simulated 32768B flash
> Using simulated 8192MB RAM
> The image is /home/lqc/xs-env/XiangShan/ready-to-run/linux.bin
> The reference model is /home/lqc/xs-env/NEMU/build/riscv64-nemu-interpreter-so
> The first instruction of core 0 has commited. Difftest enabled.
> 
> ==============  In the last commit group  ==============
> the first commit instr pc of DUT is 0x0000000080009c60
> the first commit instr pc of REF is 0x0000000080009c60
> 
> ============== Commit Group Trace (Core 0) ==============
> commit group [00]: pc 0080009c4a cmtcnt 1
> commit group [01]: pc 0080009c4e cmtcnt 1
> commit group [02]: pc 0080009c50 cmtcnt 3
> commit group [03]: pc 0080009c58 cmtcnt 3
> commit group [04]: pc 0080009c60 cmtcnt 1 <--
> commit group [05]: pc 0080009c16 cmtcnt 1
> commit group [06]: pc 0080009c1a cmtcnt 1
> commit group [07]: pc 0080009c1e cmtcnt 1
> commit group [08]: pc 0080009c22 cmtcnt 1
> commit group [09]: pc 0080009c26 cmtcnt 4
> commit group [10]: pc 0080009c32 cmtcnt 1
> commit group [11]: pc 0080009c36 cmtcnt 1
> commit group [12]: pc 0080009c3a cmtcnt 1
> commit group [13]: pc 0080009c3e cmtcnt 1
> commit group [14]: pc 0080009c42 cmtcnt 1
> commit group [15]: pc 0080009c46 cmtcnt 1
> 
> ============== Commit Instr Trace ==============
> [00] commit pc 0000000080009bfc inst 3be898f3 wen 1 dst 17 data 00003ffffffffc00 idx 020
> [01] commit pc 0000000080009c00 inst 62b89de3 wen 0 dst 27 data 0000000000000000 idx 021
> [02] commit pc 0000000080009c04 inst 0104a503 wen 1 dst 10 data 000000000000000e idx 022 (07)
> [03] commit pc 0000000080009c06 inst 00060693 wen 1 dst 13 data 0000000080046ea0 idx 023
> [04] commit pc 0000000080009c08 inst f6043c23 wen 0 dst 24 data 0000000000000000 idx 024 (02)
> [05] commit pc 0000000080009c0c inst 0015051b wen 1 dst 10 data 000000000000000f idx 025
> [06] commit pc 0000000080009c0e inst 00a4a823 wen 0 dst 16 data 0000000000000000 idx 026 (03)
> [07] commit pc 0000000080009c10 inst 0007b503 wen 1 dst 10 data 000000008000cdd8 idx 027 (08)
> [08] commit pc 0000000080009c12 inst 00068733 wen 1 dst 14 data 0000000080046ea0 idx 028
> [09] commit pc 0000000080009c16 inst 30551573 wen 1 dst 10 data 0000000080000400 idx 029
> [10] commit pc 0000000080009c1a inst 3bf028f3 wen 1 dst 17 data 0000000000000000 idx 02a
> [11] commit pc 0000000080009c1e inst 30551073 wen 0 dst 00 data 0000000000000000 idx 02b
> [12] commit pc 0000000080009c22 inst f7843683 wen 1 dst 13 data 0000000000000000 idx 02c (09)
> [13] commit pc 0000000080009c26 inst 60069ae3 wen 0 dst 21 data 0000000000000000 idx 02d
> [14] commit pc 0000000080009c2a inst 0007b503 wen 1 dst 10 data 000000008000cdd8 idx 02e (0a)
> [15] commit pc 0000000080009c2c inst 00060693 wen 1 dst 13 data 0000000080046ea0 idx 02f
> [16] commit pc 0000000080009c32 inst 30551573 wen 1 dst 10 data 0000000080000400 idx 000
> [17] commit pc 0000000080009c36 inst 3bf59073 wen 0 dst 00 data 0000000000000000 idx 001
> [18] commit pc 0000000080009c3a inst 30551073 wen 0 dst 00 data 0000000000000000 idx 002
> [19] commit pc 0000000080009c3e inst f7843683 wen 1 dst 13 data 0000000000000000 idx 003 (0b)
> [20] commit pc 0000000080009c42 inst 5e069ce3 wen 0 dst 25 data 0000000000000000 idx 004
> [21] commit pc 0000000080009c46 inst 3bf898f3 wen 1 dst 17 data 00003ffffffffc00 idx 005
> [22] commit pc 0000000080009c4a inst 5eb898e3 wen 0 dst 17 data 0000000000000000 idx 006
> [23] commit pc 0000000080009c4e inst 0104a503 wen 1 dst 10 data 000000000000000f idx 007 (0c)
> [24] commit pc 0000000080009c50 inst 00060693 wen 1 dst 13 data 0000000080046ea0 idx 008
> [25] commit pc 0000000080009c52 inst f6043c23 wen 0 dst 24 data 0000000000000000 idx 009 (04)
> [26] commit pc 0000000080009c56 inst 0015051b wen 1 dst 10 data 0000000000000010 idx 00a
> [27] commit pc 0000000080009c58 inst 00a4a823 wen 0 dst 16 data 0000000000000000 idx 00b (05)
> [28] commit pc 0000000080009c5a inst 0007b503 wen 1 dst 10 data 000000008000cdd8 idx 00c (0d)
[... 90 lines cut ...]
 cd XiangShan/
 ./build/emu -i $NOOP_HOME/ready-to-run/linux.bin
```

### Environment

- XiangShan branch: (HEAD detached at 49162c9ab)
- XiangShan commit id: 49162c9ab67070931573c1d4a372e2c858a72716
- NEMU commit id: a6a5f9b1452adef3b90a0a47f59fd38a61df1e9b
- SPIKE commit id: 
- Operating System: Ubuntu-22.04-LTS（WSL2）
- gcc version: 11.4.0
- mill version: 0.11.12
- java version: Oracle GraalVM 23.0.1+11.1

### Additional context

I am not sure whether the following information will help you understand the issue I encountered, but for the sake of readability, I am still providing it below. I executed the commands below and got the `XiangShan` Logo.
```
$ make clean
$ make riscv64-xs_defconfig
$ make -j		
$ ./build/riscv64-nemu-interpreter -b $NOOP_HOME/ready-to-run/linux.bin
```
[image]

NewPaulWalker, 2024-12-11:
I noticed that the commits you are using for XiangShan and NEMU are quite old, dating back to August. The issue with mtval has already been fixed and merged into the master. Please pull the latest XiangShan and NEMU code for testing again.

quanchenliu, 2024-12-11:
The XiangShan and NEMU branches I am using are (HEAD detached at 49162c9ab) and (HEAD detached at a6a5f9b1), respectively.  

I read the [[XiangShan documentation](https://docs.xiangshan.cc/zh-cn/latest/tools/xsenv/)](https://docs.xiangshan.cc/zh-cn/latest/tools/xsenv/), which mentions that running the `source setup.sh` command by default **tracks a stable commit on the XiangShan mainline branch, rather than the latest versions of XiangShan and other tools.** To update all submodules to their latest versions, the command `source update-submodule.sh` can be used.  Could this issue be related to the current branch I am on? 

Additionally, I could not find a branch named `HEAD` in the [XiangShan repository](https://github.com/OpenXiangShan/XiangShan/blob/master/readme.zh-cn.md). Since I have never encountered such a situation in my previous experience with Git, I have this follow-up question.

NewPaulWalker, 2024-12-12:
Okay, let me explain in more detail. 

Executing `setup.sh` updates the versions (commit hashes) of all submodules under `xs-env` to the commits tracked by `xs-env`. Currently, the `master` branch of `xs-env` tracks the versions of the XiangShan and NEMU submodules as `49162c9` and `a6a5f9b1`, which are the ones you are using now. Running `setup.sh` will not update the XiangShan and NEMU submodules to their latest branches. 

Running `update-submodule.sh` will navigate into several submodules and pull the latest code, thus updating XiangShan and NEMU to their latest versions. 

There are two things you need to note:  
1. If you run `setup.sh` again after executing `update-submodule.sh`, your XiangShan and NEMU will revert to the commits tracked by `xs-env`.  
2. XiangShan and NEMU also have their own submodules. After pulling the latest versions of XiangShan and NEMU, you need to navigate into their directories and use `git submodule` commands to ensure their submodules are also updated.

NewPaulWalker, 2024-12-12:
> Additionally, I could not find a branch named `HEAD` in the [XiangShan repository](https://github.com/OpenXiangShan/XiangShan/blob/master/readme.zh-cn.md). Since I have never encountered such a situation in my previous experience with Git, I have this follow-up question.

HEAD is not the name of any branch; it points to the commit currently in use in your repository.

If you have any further questions, feel free to ask!

quanchenliu, 2024-12-14:
Thank you for your help! I also referred to the [Git manual](https://git-scm.com/book/zh/v2) to learn more about the special branch `HEAD`. Additionally, I updated the commit versions of `XiangShan` and `NEMU`, and I am now able to successfully use `emu` and `NEMU`to execute workloads.  

I would also like to seek your advice: **after completing the above steps and becoming familiar with using AM to generate workloads, how should I continue learning the XiangShan project?**

It is worth mentioning that during the above operations, I found that **directly executing `update-submodule.sh` could not achieve the desired functionality**. However, adding two `cd ..` commands resolved the issue.

Thank you again for your assistance!

NewPaulWalker, 2024-12-16:
> I would also like to seek your advice: **after completing the above steps and becoming familiar with using AM to generate workloads, how should I continue learning the XiangShan project?**

Thank you for your interest in XiangShan. I recommend referring to this XiangShan processor getting-started guide https://docs.xiangshan.cc/zh-cn/latest/tutorials/rvsc24/. It includes presentation slides, and video recordings.

> It is worth mentioning that during the above operations, I found that **directly executing `update-submodule.sh` could not achieve the desired functionality**. However, adding two `cd ..` commands resolved the issue.

Thank you for your reminder. We noticed that this issue has already been addressed in a submitted PR, but it hasn’t been merged yet. We will proceed to merge this PR. https://github.com/OpenXiangShan/xs-env/pull/44

quanchenliu, 2024-12-17:
Okey, thanks for your help again.

closed 2024-12-17 by quanchenliu