OpenXiangShan/XiangShan #4120: Inconsistency behavior between xiangshan and NEMU after setting PMP

ha0lyu, 2025-01-02:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

After setting PMP region, inconsistency behavior occurred.

[image]

[image]

### Expected behavior

No consistence.

### To Reproduce

 _start:
   li t0, 0x70020000
    csrw pmpaddr0, t0
    li t1, 0x88
    csrw pmpcfg0, t1
    li t1, 0x1
    mret

### Environment

- XiangShan branch: master
- XiangShan commit id: 51aa1b60
- NEMU commit id: 1d750ea
- SPIKE commit id:

cebarobot, 2025-01-05:
After modifying PMP configurations, `SFENCE.VMA` and `HFENCE.GVMA` instructions are required on XiangShan, as it implements page-based virtual memory.

There is a statement in _RISC-V Privileged Architecture Manual / Machine-Level ISA / Physical Memory Protection / Physical Memory Protection and Paging_:

> ... when the PMP settings are modified, M-mode software must synchronize the PMP settings with the virtual memory system and any PMP or address-translation caches. This is accomplished by executing an SFENCE.VMA instruction with rs1=x0 and rs2=x0, after the PMP CSRs are written. ...
> 
> If page-based virtual memory is not implemented, memory accesses check the PMP settings synchronously, so no SFENCE.VMA is needed.

There is supplementary explanation in changes log about "page-based virtual memory is not implemented":

> PMP changes require an SFENCE.VMA on any hart that implements page-based virtual memory, even if VM is not currently enabled.

Tang-Haojin, 2025-01-05:
A complementary explanation: NEMU does not have detailed structures like ICache or TLB, so it is really hard to align NEMU with XS in such circumstances :(

closed 2025-09-05 by Yan-Muzi