OpenXiangShan/XiangShan #4130: 生成香山核的仿真程序时出现404

Liujiaqi-jlu, 2025-01-04:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

你好，我是按照文档的步骤来的。当我在执行make emu -j32想生成完整的香山核的时候，报错如下
[image]
不知道是什么资源访问出错，我也尝试去添加参数去打印出错的url，但是没有成功，在观察makfile和build.sbt后也没有找到结果。想请教下，这是什么原因呢。

### What did you do before

之前的操作都是按照手册来的，都已经成功，用的是v2.1版本，尚未进行nemu相关的操作

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
- mill version: Mill Build Tool version 0.11.12
- java version: 11.0.25

Tang-Haojin, 2025-01-04:
This is a known issue for older mill scripts. You may try to update your mill script from [this url](https://github.com/com-lihaoyi/mill/releases/download/0.12.5/0.12.5) and replace your mill shown by `which mill`.

Tang-Haojin, 2025-01-04:
See also: https://github.com/com-lihaoyi/mill/discussions/1482#discussioncomment-1353237.

Liujiaqi-jlu, 2025-01-04:
> This is a known issue for older mill scripts. You may try to update your mill script from [this url](https://github.com/com-lihaoyi/mill/releases/download/0.12.5/0.12.5) and replace your mill shown by `which mill`.

谢谢回复，版本问题现在已经解决了，但是在构建的时候他说找不到构建的simtop.v，请问这个要怎么解决呢
[image]

poemonsense, 2025-01-04:
v2.1 requires mill 0.9.8. Newer Mill versions have different URLs for downloading the mill tool.

poemonsense, 2025-01-04:
> > This is a known issue for older mill scripts. You may try to update your mill script from [this url](https://github.com/com-lihaoyi/mill/releases/download/0.12.5/0.12.5) and replace your mill shown by `which mill`.
> 
> 谢谢回复，版本问题现在已经解决了，但是在构建的时候他说找不到构建的simtop.v，请问这个要怎么解决呢 ![e70d2308-33c4-4dad-9122-1e776f6823e7](https://private-user-images.githubusercontent.com/142159566/400145179-c301e24f-d362-4e16-baf8-423c697fd372.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwMDE0MDksIm5iZiI6MTczNjAwMTEwOSwicGF0aCI6Ii8xNDIxNTk1NjYvNDAwMTQ1MTc5LWMzMDFlMjRmLWQzNjItNGUxNi1iYWY4LTQyM2M2OTdmZDM3Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNFQxNDMxNDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ODQ2OTI3ZWRkOTlkOGFlOTk1OTdiNzBjMmI2ZjdhNWEyM2Q3NDM4MjMyNjk3ZTlmYWFiNzcyN2Y2YTMwZjhiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.d6uNjleFiCqIoVsaOitmsHlMOGHJkQJr8UHi-ZTDdow)

You forgot to remove the `-v` argument. It is not in our script and may be added by yourself.