OpenXiangShan/XiangShan #4170: 生成波形错误

Liujiaqi-jlu, 2025-01-13:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

你好，我用的是v2.2的docker镜像版本，在生成波形的时候出现下面的报错，请问这是什么原因
[image]
他说无法打开，是有什么东西漏下了吗

### What did you do before

1

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

NewPaulWalker, 2025-01-13:
你好，我认为是你的命令输入有问题，从你输入的命令上看，emu 希望运行的 image 应该是 “ready-to-run/coremark-2-iteration.bin”，但是从你提供的报错信息上看，image 被指定为了"ready-to-run/coremark-2-iteration.bin--dump-wave"，请确保 “ready-to-run/coremark-2-iteration.bin” 和 "--dump-wave" 中间有空格

[TRANSLATION]
Hello, I think there might be an issue with your command input. Based on the command you provided, the emulator expects the image to be "ready-to-run/coremark-2-iteration.bin." However, from the error message you provided, the image is specified as "ready-to-run/coremark-2-iteration.bin--dump-wave." Please ensure there is a space between "ready-to-run/coremark-2-iteration.bin" and "--dump-wave."

Liujiaqi-jlu, 2025-01-13:
> 你好，我认为是你的命令输入有问题，从你输入的命令上看，emu 希望运行的 image 应该是 “ready-to-run/coremark-2-iteration.bin”，但是从你提供的报错信息上看，image 被指定为了"ready-to-run/coremark-2-iteration.bin--dump-wave"，请确保 [...]

谢谢，确实是命令的问题，现在已经解决。但是有下面的问题
[image]
我看了一下他提到的这个动态库是在ready-to-run这个文件夹下面的
[image]

NewPaulWalker, 2025-01-13:
riscv64-nemu-interpreter-so 是用来和 XiangShan 仿真程序 emu 一起做差分测试的动态连接库。你可以在运行 emu 时指定 “--no-diff” 来关闭差分测试，或者 “--diff=PATH” 来指定动态链接库的路径。更多的用法可以运行 “./build/emu -h”来查看。这里因为没有显示指定动态链接库的路径，所以 emu 尝试了一个可能的路径，但是在该路径下没有对应的库文件，因此发生了错误
[TRANSLATION]
The `riscv64-nemu-interpreter-so` is a dynamically linked library used for differential testing with the XiangShan simulation program `emu`. You can disable differential testing by specifying `--no-diff` when running `emu`, or use `--diff=PATH` to specify the path to the dynamic library. For more usage options, you can run `./build/emu -h` to check.
In this case, since the path to the dynamic library was not explicitly specified, `emu` attempted a possible path but couldn't find the corresponding library file at that location.

NewPaulWalker, 2025-01-13:
Please try 
`./build/emu -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so --dump-wave -b 10000 -e 11000` 
or 
`./build/emu -i ./ready-to-run/coremark-2-iteration.bin --no-diff --dump-wave -b 10000 -e 11000`

Liujiaqi-jlu, 2025-01-13:
> Please try `./build/emu -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so --dump-wave -b 10000 -e 11000` or `./build/emu -i ./ready-to-run/coremark-2-iteration.bin --no-diff --dump-wave -b 10000 -e 11000`

好的谢谢，问题已经解决

closed 2025-04-24 by Ma-YX