OpenXiangShan/XiangShan #4199: When specifying -march=rv64gc_zba_zbb_zbc_zbs to generate checkpoints, the checkpoints still contain vector instructions

zybzzz, 2025-01-18:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

## question

I used ```deterload``` to generate checkpoints for Xiangshan execution. During the checkpoint generation, I explicitly specified the compilation option as ```-march=rv64gc_zba_zbb_zbc_zbs```.

```nix
  export LiteWrapper=$(realpath ${CPU2006LiteWrapper.name})
  export SPEC=$(realpath ./spec2006)
  cd $LiteWrapper

  export SPEC_LITE=$PWD
  export ARCH=riscv64
  export CROSS_COMPILE=riscv64-unknown-linux-gnu-
  export OPTIMIZE="-O3 -flto -march=rv64gc_zba_zbb_zbc_zbs"
  export SUBPROCESS_NUM=5

  export CFLAGS="$CFLAGS -static -Wno-format-security -I${customJemalloc}/include "
  export CXXFLAGS="$CXXFLAGS -static -Wno-format-security -I${customJemalloc}/include"
  export LDFLAGS="$LDFLAGS -static -ljemalloc -L${customJemalloc}/lib"

  pushd $SPEC && source shrc && popd

```

**I did not expect the generated checkpoints to contain vector instructions.** However, when running the corresponding checkpoint on xs-gem5, the statistics and runtime information from xs-gem5 indicated that a small number of related instructions were still executed.

```txt
system.cpu.commit.vectorInstructions             1056                       # Number of committed Vector instructions. (Count)
system.cpu.statIssuedInstType_0::VectorUnitStrideLoad          523      0.00%    100.00% # Number of instructions issued per FU type, per thread (Count)
system.cpu.statIssuedInstType_0::VectorSegUnitStrideLoad            0      0.00%    100.00% # Number of instructions issued per FU type, per thread (Count)
system.cpu.statIssuedInstType_0::VectorUnitStrideStore          518      0.00%    100.00% # Number of instructions issued per FU type, per thread (Count)

```

I am trying to figure out why this is happening. Could it be that during the static linking process for checkpoint generation, some libraries containing vector instructions were linked? I am puzzled about this issue.

## env

- deterload version: d2fc4767eb72d6922418003f013687e8d3d6349f

eastonman, 2025-01-18:
If you are using DeterLoad, this is expected.

With new Linux kernel and GLIBC, GLIBC can detect which ISA is supported via the `hw_probe` syscall. In DeterLoad, the simulator used (QEMU/NEMU) enables V. The optimization flags specified here only apply to user programs like SPEC CPU. GLIBC is pre-compiled by Nix and supports V. When a user program calls GLIBC functions, GLIBC uses **ifunc** to dynamically dispatch to a V path (if that function has a V path). This results in a small amount of vector instructions being executed on DeterLoad checkpoints.

eastonman, 2025-01-18:
I believe this is a DeterLoad issue. There exists methods that can totally avoid vector instruction, like modifying device tree to tell kernel there are no V.

Maybe we should open an issue in DeterLoad and discuss with @xieby1

zybzzz, 2025-01-19:
@eastonman Thank you for your response. I have a general understanding of the possible causes of this issue, and I have the following two thoughts:

**Dynamic monitoring via ifunc:** It seems that ifunc dynamically monitors hardware support for ISA extensions at runtime. This implies that, to avoid the aforementioned issue, we need to completely disable runtime simulator support for specific ISA extensions. For instance, we might need to ensure that the V extension is entirely disabled in NEMU throughout the runtime.

**The role of ld.so:** From my brief research, I discovered that ifunc actually relies on the ld.so dynamic linker. However, based on my understanding, the initramfs used to generate the checkpoints does not seem to include ld.so in our Linux system image. This suggests that when generating the checkpoints, there might not be a usable ld.so in the system. Since the ifunc mechanism depends on ld.so, I am puzzled about how this apparent contradiction arises.

I admit that my understanding of the latest kernel technologies and GLIBC's ifunc mechanism is limited. My two thoughts above may contain misunderstandings or incomplete information. I think I need to take some time to learn more about these related technologies.

eastonman, 2025-01-19:
Not sure about ld.so and glibc dynamic linking. I have not digged into the root cause yet. The aforementioned reason is just a guess.

Maybe ask some experts in ifunc @cyyself.

closed 2025-09-10 by Yan-Muzi