OpenXiangShan/XiangShan #4240: CMO Extension, `cbo.zero` instruction details

plutsyk, 2025-01-27:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

It seems that implementation of the `cbo.zero` instruction would require updates to few other places in the code base, analogous to those made at [StoreQueueData.scala#L78](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/mem/lsqueue/StoreQueueData.scala#L78). For instance, in case of [LoadQueueRAW.scala#L287](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/mem/lsqueue/LoadQueueRAW.scala#L287), it would require checks whether "querying" stores (in `io.storeIn`) are coming from the `cbo.zero` instruction, i.e. writing the entire cache lines or not. Currently such checks seem to be missing, which strongly suggests possible issues for memory model correctness... unless this is taken care of elsewhere in the design. Perhaps you could provide a bit of clarification or hints in the latter case.

Anzooooo, 2025-01-28:
Yes, this does present a certain problem, thank you for pointing this out and we will address this issue subsequently.

Anzooooo, 2025-02-18:
The issue has now been resolved by the pr mentioned, thanks for pointing it out.