OpenXiangShan/XiangShan #4242: Possible bug in statistical corrector

castleberrysam, 2025-02-02:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

Hello, while inspecting code for the statistical corrector (SC.scala), I noticed a possible bug in the write conflict and forwarding mechanism.

Background: When there is a read-write conflict to the table (read and write to the same index in the same cycle), the read takes priority, and the write data is instead saved to conflict_buffer until it can be written to the table in a later cycle. When there is a read to the same index while the conflict_buffer is valid (including the read of the original read-write conflict), then the read result is forwarded from the conflict_buffer, since the value in the table is stale.

Problem: The table contains four ways per entry, but only one of the ways is changed during a write. So when forwarding from the conflict_buffer, we should expect that one of the ways is from the conflict_buffer, and the other three ways are from the table. However, it seems that in the current logic, when forwarding from the conflict_buffer, the other three ways are always zero. I also looked at a waveform from microbench to confirm this behavior.

[image]

At marker 0: Read at index 84, the read result is {02, 05, 00, 00}
At marker 1: Read-write conflict at index 84 way 1, the read result is {00, 06, 00 00}
At marker 2: Read at index 84, the read result is {02, 06, 00, 00}
-> I expected that the read result at marker 1 would be {02, 06, 00, 00}.

However, since I am not very familiar with the XiangShan design, I am not sure whether this behavior will cause any issue downstream. If this is a real bug, I can try to write a fix.

The conflict_buffer logic was added in #3671.

Thanks, Sam

### Expected behavior

During read-write conflict to the statistical corrector table, the read result should match the case where the write occurred before the read. In other words, the read result for the write way should be forwarded from the write data, and the read result for the other ways should be from the table.

### To Reproduce

Steps:
1. Gather waveform for a workload with non-trivial branch activity, I used microbench from ready-to-run folder.
2. Locate read-write conflict to SC table with use_conflict_data signal in SCTable, the entry should also have more than one non-zero way.
3. Observe that the read result of the read-write conflict contains zero for all ways other than the write way.
4. Observe that reads to this index following the read-write conflict have the correct value. (The problem is only present during write forwarding.)

### Environment

- XiangShan branch: master
- XiangShan commit id: 977ac3b188267f31bfb32c7fa358f8b923a0e985
- NEMU commit id: N/A
- SPIKE commit id: N/A

eastonman, 2025-02-13:
Thank you for your report. I can confirm that this is not expected and needs to be fixed. Feel free to submit a fix, as it would be greatly appreciated.

Apologize for the late response.

eastonman, 2025-02-13:
Also, you may want to checkout these codes.
https://github.com/OpenXiangShan/Utility/blob/6ed7394ef85887ea360f33b3be97bb13fb31e57c/src/main/scala/utility/SRAMTemplate.scala#L301-L358

We have tried moving the conflict management code to the SRAM template. I believe this approach is better, as every dual-port SRAM requires it if the memory compiler does not support same-address read-write operations.

castleberrysam, 2025-02-14:
Okay understood, thanks for the advice. I'm travelling until late next week, I will try to make a fix for it then.

castleberrysam, 2025-02-26:
Hello, I'm looking into adding this function to the RAM wrapper now.

The current logic in SC blocks the RAM write during a conflict. However, at least for dual-port RAMs I have used in the past, this would not be necessary. When those RAMs received read and write at the same index, the write would still succeed, but the read result would be corrupt for ways that were written. In this case it is only necessary to bypass the write data, not block the write operation. It is better than the current logic, since the current logic may drop writes when there are back-to-back conflicting reads every cycle.

My question is, what is the required behavior for the dual-port RAM macros? Given that different RAM macros may have different behavior, it is a trade-off between performance and compatibility. I see three possible answers:
1. Simultaneous read and write to same index allowed, but read result is corrupt for ways that were written
2. Simultaneous read and write to same index allowed, but read result is corrupt for all ways
3. Simultaneous read and write to same index not allowed

From a design perspective, I would like to assume (1), since this is best for performance, and RAM macros I used in the past support this. But it would also be possible to make this configurable, in case a future user of XiangShan cannot use RAM macros with this behavior. Please let me know what the preferred choice is for XiangShan. Thanks, Sam

eastonman, 2025-03-19:
Any progress on this issue? @castleberrysam 
I have received user feedback saying that the previous fix https://github.com/OpenXiangShan/XiangShan/pull/3671 on SC is broken, and still trying to read and write to the same index. I will be looking into this issue in one or two days. I am here to ask if you have done any work. If so I will try not to waste it.

castleberrysam, 2025-03-20:
Yes, I have some code to add a few conflict handling modes to the SRAMTemplate. I also made some directed tests to make sure it works. I put it aside since I've been busy job hunting and felt it still needs more testing. But I can ignore that for now and put it into a pull request for review.

eastonman, 2025-03-20:
Oh that's really appreciated ☺️

castleberrysam, 2025-03-21:
Hello, I submitted two pull requests which should be shown above.

eastonman, 2025-03-21:
Wow, remarkable work. To be honest I have not seen such high-quality contributions before. I will start reviewing these codes and testing them in my environment ASAP.

eastonman, 2025-03-25:
I've added a comment on your SRAMTemplate PR https://github.com/OpenXiangShan/Utility/pull/110#issuecomment-2747211940
In case you miss it.