OpenXiangShan/XiangShan #4270: make simv RELEASE=1 :  compile-error

XIAOMILeoWang, 2025-02-14:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

[image]

### What did you do before

I've installed VCS2018.

### Environment

- XiangShan branch: master
- XiangShan commit id: 49162c9ab67070931573c1d4a372e2c858a72716 XiangShan (v0.2020-5724-g49162c9ab)
- NEMU commit id: a6a5f9b1452adef3b90a0a47f59fd38a61df1e9b NEMU (remotes/origin/tmp-merge2-77-ga6a5f9b1)
- SPIKE commit id: b03c55a5df5dc8793cb44b42dd60141566e57e78 ./XiangShan/rocket-chip/scripts/arch-test/spike?
- Operating System: Ubuntu 20.04.6 LTS
- gcc version: 9.4.0
- mill version: 0.11.8
- java version: 1.8.0_25

Tang-Haojin, 2025-02-17:
It seems that your compiling fails at very beginning at the step of mill. Maybe this should be a mill issue.

Tang-Haojin, 2025-02-17:
You can try to remove `out` directory under XiangShan and try again.

XIAOMILeoWang, 2025-02-18:
Thanks a lot, bro!!!  It works after I removed out dir.  
I've googled it and it seems like a bug in mill. The bug has been fixed after mill 0.12.2.(https://github.com/com-lihaoyi/mill/issues/3874)

Thanks again for your help.

closed 2025-02-18 by XIAOMILeoWang