OpenXiangShan/XiangShan #4271: XSCore.scala:66:33: reference to MemBlock is ambiguous;

zxl819, 2025-02-14:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

[image]
XSCore.scala:66:33: reference to MemBlock is ambiguous;

### Expected behavior

为什么make verilog中xscore.scala文件报错？在运行前我已经更新了子模块。

[image]

### To Reproduce

在Xiangshan目录下make verilog

### Environment

- XiangShan branch:origin/master
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id:

Tang-Haojin, 2025-02-14:
Try to `rm -rf out` at XiangShan directory and try again please :)

zxl819, 2025-02-14:
好的我试一下

zxl819, 2025-02-14:
> Try to `rm -rf out` at XiangShan directory and try again please :)
感谢，我尝试了重装。

closed 2025-02-14 by zxl819