OpenXiangShan/XiangShan #4276: clang-14: error: clang frontend command failed with exit code 139 (use -v to see invocation)

zxl819, 2025-02-14:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

[image]
请问这个问题再make emu CONFIG=MinimalConfig 的时候出现是为什么？

### Expected behavior

正确编译emu

### To Reproduce

make emu CONFIG=MinimalConfig

### Environment

- XiangShan branch: origin/main
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id:
-

Tang-Haojin, 2025-02-14:
According to InstallDir in your screenshot, it may be your environment problem.

zxl819, 2025-02-14:
> According to InstallDir in your screenshot, it may be your environment problem.

Did I install in a wrong dir？
我应该安装在哪类文件夹呢？

Tang-Haojin, 2025-02-14:
You should not install it manually. Your environment on open03 already has clang installed. For environmental problems on BOSC machine, it may be more convenient for you to consult your colleagues :)

closed 2025-02-15 by zxl819