OpenXiangShan/XiangShan #4278: ALU.scala

mlabaf2, 2025-02-16:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

Hi I added new outputs to class FuOutput to use in Alu.scala.
but I cannot find instance of class Alu in alu.scala . I want to connect some new outputs of Alu class to another module.

Thanks

### What did you do before

Hi I added new outputs to class FuOutput to use in Alu.scala.
but I cannot find instance of class Alu in alu.scala . I want to connect some new outputs of Alu class to another module.

Thanks

### Environment

- XiangShan branch: a455b9f714df264edb679c81c3d6c74625ea3d82
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Tang-Haojin, 2025-02-17:
For instantiating of Alu, you can refer to `AluCfg` in `FuConfig.scala`. The `fuGen` of case class FuConfig describes how a function unit should be instantiated, and the actual instantiating takes place at https://github.com/OpenXiangShan/XiangShan/blob/2fa6e783e0d46a33b864707ca720b7408c628a16/src/main/scala/xiangshan/backend/exu/ExeUnit.scala#L67

mlabaf2, 2025-02-19:
thanks for response.
I added some new in and out in FuOutput and FunctionUnitInput to have newer in/out in class Alu but I do not have access to them in Exe.scala to connect these signals to other module. Maybe I should add new alu in/out to ExuInput and ExuOutput in bundle.scala. what is the difference between them? 
thanks

Tang-Haojin, 2025-02-19:
An ExeUnit contains one or more than one FunctionUnit, so their IO bundles also differ. That means, if you want to expose some new IOs from FU to somewhere like LSU, you should add your new IOs to IO bundles in FU, ExeUnit, ExuBlock, etc. and connect them correctly. You should "bore" your new IOs across multiple module hierarchy to connect them to outer modules. For more details, please read XiangShan source code.

The commit module you referred to may be Rob in XiangShan. Rob is the place where XiangShan actually does instruction retiring.

mlabaf2, 2025-02-25:
Thank you for responses.
what is the name if issue module?
where is the source of input signals of Rob and where is the instance of Rob?
I want to connect extra signals between Rob, CSR, Refile, Exu module. Is there any main module that consists instance of these module same as other usual CPU...

Tang-Haojin, 2025-02-25:
For issue module, you can see `Scheduler` and `IssueQueue`.

All these modules you mentioned are in a module (or submodules of that) named `Backend`. Have you ever tried generate verilog files of XiangShan? If you are not familiar with scala, you can see generated verilog files.

mlabaf2, 2025-02-25:
> For issue module, you can see `Scheduler` and `IssueQueue`.
> 
> All these modules you mentioned are in a module (or submodules of that) named `Backend`. Have you ever tried generate verilog files of XiangShan? If you are not familiar with scala, you can see generated verilog files.

Thanks
In Yanqihu version, none of the  `Scheduler` and `IssueQueue` exist. also backend.....

Tang-Haojin, 2025-02-25:
We no longer maintain yanqihu actually. Please consider using the more stable kunminghu (master) or nanhu instead.

closed 2025-09-05 by Yan-Muzi