OpenXiangShan/XiangShan #4289: Troubles of FPGA prototyping on Xilinx xcvu19p

San0519, 2025-02-19:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

I am doing FPGA prototype on Xilinx xcvu19p. I've followed the steps given by the tutorial document in "FPGA Prototype" section, but the linux kernel seems not work still. One step in this section metioned the tcl would catch and print an error message during writing the DDR. I am not sure whether the image was correctly loaded into DDR, though I've seen literally a error message when running the tcl, which was about AXI fails and gave the "WRITE DATA". Could you provided any further details during this step or about FPGA prototype, or the actual error message type. The bitstream and ltx file were generated by Vivado 2020.2 and programed to FPGA by Vivado 2023.2.

lewislzh, 2025-02-20:
Hi, thanks for your issue. 
Could you please provide the XiangShan version you are using?

San0519, 2025-02-20:
> Hi, thanks for your issue. Could you please provide the XiangShan version you are using?
Thanks for your response,
I cloned the project from the Nanhu branch with id 09d2a32142c64fca898e17c0b943e61ddc331958 that was provided in the tutorial document, but I made verilog with single core version with "make verilog #NUM_CORES=1"

San0519, 2025-02-20:
Thanks for your response,
I cloned the project from the Nanhu branch with id https://github.com/OpenXiangShan/XiangShan/commit/09d2a32142c64fca898e17c0b943e61ddc331958 that was provided in the tutorial document, but I made verilog with single core version with "make verilog #NUM_CORES=1"

xianfeili, 2025-02-20:
Could you provide the program log and the uart output log. So we can try to locate the root cause. For the dual-core image can also workwith the single core. "make verilog  #NUM_CORES=1" this should not be the fundamental issue.

San0519, 2025-02-20:
Sure, the log output during programming FPGA by running the tcl file was
`
IRNING: [Labtoolstcl 44-529] The argument '-size' is deprecated. The specified value, '32', will be ignored. The data word size will be automatically set to '32' based on the IP core configuration
RROR:
[Xicom 50-38] xicom: AXI TRANSACTION FAILED
ERROR:
[Labtoolstcl 44-481] WRITE DATA is: {The First 2048 hex bit of provided "data.txt“ file}
ErrorMsg: ERROR: [Common 17-39] "run_hw _axi' failed due to earlier errors.
puts
"After Error"
`
Here was the main problem that I am not sure this error message was supposed to occur or align with the mentioned one in the document. In addition, the part of DDR was modified to match the source on the target device, I am wondering if there any possibility that the changed DDR was incompatible that caused the writing failure.

San0519, 2025-02-20:
Sure, the log output during programming FPGA by running the tcl file was
`WARNING: [Labtoolstcl 44-529] The argument '-size' is deprecated. The specified value, '32', will be ignored. The data word size will be automatically set to '32' based on the IP core configuration`
`ERROR: [Xicom 50-38] xicom: AXI TRANSACTION FAILED ERROR: `
`[Labtoolstcl 44-481] WRITE DATA is: {The First 2048 hex bit of provided "data.txt“ file}` 
`ErrorMsg: ERROR: [Common 17-39] "run_hw _axi' failed due to earlier errors. `
`puts "After Error"`
Here was the main problem that I am not sure this error message was supposed to occur or align with the mentioned one in the document. In addition, the part of DDR was modified to match the source on the target device, I am wondering if there any possibility that the changed DDR was incompatible that caused the writing failure.

xianfeili, 2025-02-20:
En~~~The last code that before the "WARNING: [Labtoolstcl 44-529] The argument '-size' is deprecated. The specified value, '32', will be ignored. The data word size will be automatically set to '32' based on the IP core configuration" is a all zero write data ? If the answer is yes. Seems the program step is OK. If the error message happened at the beginning or middle of the data.txt download step. Need check the JTAG speed and connect. Try lower speed program may solve the problem.

San0519, 2025-02-20:
Did you mean the begin of target written address of DDR? The data before you mentioned was actually a zero. Is that correct to think the error message during programming device was normal under this situation? BTW, I will try a lower speed both on my DDR and JTAG comm. 
Thanks for all your help so far, I will let you know if it works.

San0519, 2025-02-21:
I tried a lower JTAG comm frequency from default 15M to 10M, however the results seemed the same at all. I inserted ILA attached to AXI interface, and the target "write data" wires only triggered once when running the tcl, so did the printed message that I added within the loop. The while loop of writing data.txt file was interrupted by the error message catching block. I thought it was caused by the DDR incompatibility. Also, could you please tell me why the tcl only put the first character of $AddrString and $DataString but not the whole line $aline and $dline? I have no experience on tcl scripts but it did choke me.

xianfeili, 2025-02-21:
Two things need to do. 1. open your vivado hardware manager check the MIG status is fine. "CAL PASS" 2. use JTAG AXI master manually.

San0519, 2025-02-21:
I've re-run the tcl after increasing the time delay before writing the DDR, and the data and address read back was correctly. Appreciate your detailed and immediate help.

closed 2025-02-21 by San0519