OpenXiangShan/XiangShan #4294: csr

mlabaf2, 2025-02-20:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hi
I want to write or read content of csrs (mie,mip, mstatus, ....) when an activation signal is '1'. I added these three IO ( enable, data_in , data-out)  to CSRFileIO. module. Now how can I update content of CSRs by data-in when enable is '1'? How can I write something to CSRs? 
also I want to update the content of GPRs in a same manner. I cannot fine where are GPRs defined in XiangShan and how can I update them? 
thank for support

NewPaulWalker, 2025-02-20:
> I want to write or read content of csrs (mie,mip, mstatus, ....) when an activation signal is '1'. I added these three IO ( enable, data_in , data-out) to CSRFileIO. module. Now how can I update content of CSRs by data-in when enable is '1'? How can I write something to CSRs?

Regarding what you want to do, I think you can refer to the implementation of some custom CSRs in XiangShan. For example, the `l2FlushDone` field of the custom CSR `mflushpwr` is read-only and directly driven by a signal from another module. You could consider applying a similar approach to the CSRs you want to modify the read/write behavior for.

> also I want to update the content of GPRs (in integer part/alu) in a same manner. I cannot find where are GPRs defined in XiangShan and how can I update them?

As for what you want to do, I don’t have much of an idea, because it seems like this could be accomplished directly with an instruction like `li a0, 0xabcdef`. Also, our general-purpose registers are located in the datapath module, and it’s called `Regfile`.

NewPaulWalker, 2025-02-20:
> I am using a455b9f714df264edb679c81c3d6c74625ea3d82 commit version of yanqihu. there is not `l2FlushDone` or `mflushpwr` to find out the way.....

Sorry, we no longer maintain yanqihu. Please use the more stable kunminghu (master) instead.

closed 2025-09-05 by Yan-Muzi