OpenXiangShan/XiangShan #4323: Phenomenon after loading mirroring

San0519, 2025-02-27:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

I am now trying to loading a distro on my FPGA, however, the hardware peripherals are limited to loading by USB disk. To ensure all the previous steps were conducted correctly, I am wondering if there is any printed information or other expected results after loading the provided *data.txt* file without USB booting.

Ma-YX, 2025-02-27:
Hi, thanks for your issue.
Could you please provide the XiangShan version you are using, and the program log together with the uart output log?

San0519, 2025-02-27:
Thanks for your response, I cloned the project from the Nanhu branch with id https://github.com/OpenXiangShan/XiangShan/commit/09d2a32142c64fca898e17c0b943e61ddc331958 that was provided in the tutorial document on FPGA xcvu19p. The uart output was only 'T O D' and 54 4F 44 in hexadecimal
The log was
```
INFO: [Common 17-14] Message 'Labtoolstcl 44-481'
appears 100 times and further instances of the messages will be disabled.
ERROR: [Xicom 50-38] xicom: Internal error: bad hexStr for Cse NumberUtil: :hexStr2BinStr
Use the Tcl command set msg config to change the current settings.
ErrorMsg: ERROR: [Common 17-39] ' run_hw_axi failed due to earlier errors.
#puts "After Error"
#After Error
#startgroup
#set_property OUTPUT_VALUE 0 [get hw probes vio_sw5 -of_objects [get hw_vios -of_objects lget_hw _devices xcvu19p_3] -filter {CELL_NAME=~"u vio"}]] commit_hw _vio [get_hw probes {vio_sws) -of _objects lget_hw_vios -of _objects [get_hw devices xcvu19p_3] - filter [CELL_NAME=~ "u_vio"}
#endgroup 
#after 5000 
#startgroup
#set_property OUTPUT_ VALUE 1 [get_hw probes vio_sw -of_objects [get_hw_vios -of_objects [get_hw _devices xcvu19p_3] -filter {CELL_NAME=~ "u vio"}l commit_hw_vio [get _hw probes (vio_sw5} -of_objects lget hw_vios -of_objects [get_hw devices xcvu19p_3] -filter [CELL_NAME=~"u_vio"}]}
#endgroup
# after 5000
```

xianfeili, 2025-03-04:
[nanhu_boot.log](https://github.com/user-attachments/files/19062511/nanhu_boot.log)
Here is the boot log I try on my own Env. Can you provide the clk you use on your board ?

San0519, 2025-03-04:
I had a 100MHz clock, and divided into 80MHz for DDR, 200MHz for system clock and a 50MHz clock for debug core.

xianfeili, 2025-03-04:
There is no difference between our clocks. Is there any modified when porting our design into yours ?

xianfeili, 2025-03-26:
I've try two things these weeks.

1. Generate the single core version with "make verilog #NUM_CORES=1". It can work with the vivado project and data.txt SW.
2. Use the 50MHz board clock to generate the 200MHz, 50MHz and 10MHz(replace the 1MHz clock). It also works well with the single core version.

So, I've no idea why it can not work on your platform

San0519, 2025-03-27:
Thanks for your effort on this issue. I've modified the reset part for this project, I guess that is what caused the problem. Due to the limit in the physical interface on my platform, I cannot reproduce the whole process provided in your document. Again, sincerely appreciate for your reply.

closed 2025-03-27 by San0519