OpenXiangShan/XiangShan #4327: Problem with generating verilog

SalaoNomii, 2025-02-27:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

在运行了make verilog命令之后，在生成的build/rtl文件夹下我找不到backend.sv这个文件

### What did you do before

我按照香山官网前端开发环境流程，通过make verilog跑出了 build文件夹

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Tang-Haojin, 2025-02-27:
Please provide your corresponding Environment and your error message if any.

Tang-Haojin, 2025-02-27:
Once the required information is provided, please reopen this issue.