OpenXiangShan/XiangShan #4331: Expandding data bus

mlabaf2, 2025-02-27:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hi 
I want to expand GPRs , Data bus, Data cache to 128 bit , and all related parameters, I changed XLEN to 128 but received error, what are the other parameters that should be change to be able to read and write 128 bit to and from memory? 
I cannot find source of this error....

[image]

Thanks

Tang-Haojin, 2025-02-28:
Thanks for your question. XLEN is not designed as a variable but a constant. We basically do not allow users to change this parameter. If you really want to change it, a way is to read the error message carefully because this will tell you why the compilation fails directly. After all compilation errors are fixed, you can try your self-written workload which is compatible for XLEN=128 and debug via waveform.

Tang-Haojin, 2025-02-28:
For the error message you provided, I suggest you to look from PMP.scala because this is the last file we written shown in this error message.

Tang-Haojin, 2025-02-28:
We may publish our design documents shortly, but IO connection may not be available in the first edition.

Tang-Haojin, 2025-02-28:
And I actually wonder why you want to change XLEN to 128.

mlabaf2, 2025-02-28:
> We may publish our design documents shortly, but IO connection may not be available in the first edition.

Is IO connection available for previous version of XiangShan?

mlabaf2, 2025-02-28:
Thanks for your response. Is it possible to have access to architectural register in XiangShan instead of physical register?

Tang-Haojin, 2025-03-01:
> Thanks for your response. Is it possible to have access to architectural register in XiangShan instead of physical register?

We have an `ArchTable` in kunminghu (master), so you can use register mapping in `ArchTable` to find the corresponding physical register in register file to get architectural register of committed state. But in Nanhu, this structure does not exist.

Tang-Haojin, 2025-03-01:
> > We may publish our design documents shortly, but IO connection may not be available in the first edition.
> 
> Is IO connection available for previous version of XiangShan?

We have IO port explanations of some relatively important and large modules of Nanhu. If you are looking at Nanhu, you can tell me which module(s) you are interested in.

mlabaf2, 2025-03-02:
> We have IO port explanations of some relatively important and large modules of Nanhu. If you are looking at Nanhu, you can tell me which module(s) you are interested in.

Please Rob.scala , csr.scala, regfile.scala, exeblock.scala,  ctrlblock.scala

thanks