OpenXiangShan/XiangShan #4342: 上板测试最小化xiangshan工程跑完onboard-ai1-119.tcl脚本串口只打印了几个字符使什么原因

wj-online, 2025-03-01:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

生成了一版最小化xiangshan的FPGA工程，程序改写了DDR型号，下载到xcvu19p的板子上，跑完onboard-ai1-119.tc脚本后串口只打印了“tod”这几个字符就没有了，DDR内的data.txt经过测试是写进去了的，不知道还有那个流程不对，CPU复位后没有DDR也没有读的操作。

### What did you do before

尝试了多次复位CPU，每次都只打印了几个相同字符，程序内改写外部复位无效化，外部时钟本地生成。

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

lewislzh, 2025-03-02:
Hi, thanks for your issue.
Could you please provide the XiangShan version you are using?

xianfeili, 2025-03-03:
Which characters it print ? If uart have output, it means boot-code is working. CPU is jump to run the code download into the DDR.

wj-online, 2025-05-13:
our fpga prototye in nanhu 2core, branh is nanhu_clockdiv2, commit id is: [09d2a32142c64fca898e17c0b943e61ddc331958](https://github.com/OpenXiangShan/XiangShan/commit/09d2a32142c64fca898e17c0b943e61ddc331958) we have some problem (we changed the DDR type and clock, use vivado ip to generate clock).
uart0 print three letter "TOD" and stop,SOC_M_AXI read just one time and stop , the data we checked is right, still don't know why cpu stopped axi read action.

xianfeili, 2025-05-16:
Here is the ila waveform that we capture on our own platform during the boot. You can check the different.

[nanhu_ila.zip](https://github.com/user-attachments/files/20237186/nanhu_ila.zip)

wj-online, 2025-05-19:
I changed parameter in  AXI_interconnect of jtag_ddr_subsystem ,soc_m_axi can read to "0x0000_2cc0" and stopped , uart output "60 66 18 20"four byte message , is this relate to timing or clock?
otherwize,can i have your project for tert?

wj-online, 2025-05-19:
I changed Data width of the axi crossbar from 32 to 64, S01AXI Arbitration Priority to "1".