OpenXiangShan/XiangShan #4343: trouble with nanhu prototype board test

wj-online, 2025-03-01:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

I build a one core nanhu prototype on xilinx xcvu19p, we finished all step in onboard-ai1-119.tcl，all tcl step seems right .but there's just few data out from uart,  seems linux did't boot or work.

### What did you do before

I try to reset cpu many times , Same phenomenon

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

lewislzh, 2025-03-02:
Hi, thanks for your issue. 
Could you please provide the specific XiangShan version you are using?

wj-online, 2025-03-03:
I cloned the project from the Nanhu branch with id https://github.com/OpenXiangShan/XiangShan/commit/09d2a32142c64fca898e17c0b943e61ddc331958 that was provided in the tutorial document, but I made verilog with single core version with "make verilog #NUM_CORES=1"

xianfeili, 2025-03-03:
Can you provide the uart output log so that we can try to locate the root cause.

wj-online, 2025-03-03:
I use Serial Debugging Assistant monitor the uart output , it output  three letter word "tod" and end.

xianfeili, 2025-03-04:
Can you provide the clock you use on your own board ?

xianfeili, 2025-03-26:
Sorry  for late response. I've try two things these weeks.
1. Generate the single core version with "make verilog #NUM_CORES=1". It can work with the vivado project and data.txt SW.
2. Use the 50MHz board clock to generate the 200MHz, 50MHz and 10MHz(replace the 1MHz clock). It also works well with the single core version.

So, I've no idea why it can not work on your platform

closed 2025-05-28 by wj-online