OpenXiangShan/XiangShan #4365: 关于workload在AM裸机环境下的地址映射问题

Liujiaqi-jlu, 2025-03-05:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

你好，我在AM裸机环境下编译好了workload，并且在香山上运行，出于研究的需要，我想请问一下，workload中的逻辑地址与香山处理器中的物理地址存在什么映射关系，在裸机环境下是直接把逻辑地址用作物理地址嘛？期待回复！谢谢！

cebarobot, 2025-03-05:
在裸机环境下，若无特殊配置，取指和访存均直接按物理地址进行。若希望启用分页虚拟地址系统，则需要根据 RISC-V 特权手册的要求配置 satp 等 CSR 寄存器、在内存中配置页表、进入 S 或 U 特权模式。参见 [riscv/riscv-isa-manual](https://github.com/riscv/riscv-isa-manual)。

对于 Workload，香山仿真环境会直接将工作负载文件按原样加载到物理地址为 0x80000000 的内存空间中。参见 [工作负载概述](https://docs.xiangshan.cc/zh-cn/latest/workloads/overview/)。

Liujiaqi-jlu, 2025-03-05:
> 在裸机环境下，若无特殊配置，取指和访存均直接按物理地址进行。若希望启用分页虚拟地址系统，则需要根据 RISC-V 特权手册的要求配置 satp 等 CSR 寄存器、在内存中配置页表、进入 S 或 U 特权模式。参见 [riscv/riscv-isa-manual](https://github.com/riscv/riscv-isa-manual)。
>
> 对于 [...]

请问对于workload中的读写变量过程相对于处理器的AXI地址也是同样的直接用逻辑地址用作物理地址是吧

cebarobot, 2025-03-05:
> 请问对于workload中的读写变量过程相对于处理器的AXI地址也是同样的直接用逻辑地址用作物理地址是吧

读写变量即访存。对于一般的 Nexus-AM 程序，是的，程序中引用的逻辑地址即是物理地址。

Liujiaqi-jlu, 2025-03-05:
> > 请问对于workload中的读写变量过程相对于处理器的AXI地址也是同样的直接用逻辑地址用作物理地址是吧
> 
> 读写变量即访存。对于一般的 Nexus-AM 程序，是的，程序中引用的逻辑地址即是物理地址。

好的，谢谢解答！

cebarobot, 2025-03-05:
如果还有其他问题，请随时创建新的 Issue。