OpenXiangShan/XiangShan #4368: Vector Configuration

WaltJr-OH, 2025-03-06:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hi, I have a question about risc-v vector, can i configure the VLEN and how many valus are implemented in xiangshan? is it 2*128? is it configurable.

Tang-Haojin, 2025-03-06:
Thanks for your question. Only VLEN=128 is currently supported by XiangShan.

SalaoNomii, 2025-03-07:
> Thanks for your question. Only VLEN=128 is currently supported by XiangShan.
Thanks for responding. I tried some simple running cases, i improved arithmetic intensity from 2 to 256. It seems the peak performance is closely under 4 32 elements per cycle (vadd instruction). In the micro-architecture image, I saw 2 vialu. I have no clue what happens

WaltJr-OH, 2025-03-07:
> Thanks for your question. Only VLEN=128 is currently supported by XiangShan.

Thanks for responding, one of my colleagues is running vector performance experiment, and it seems the peak performance can never reach 4*32 elements per cycle (vadd). We are wondering what happens

camel-cdr, 2025-03-09:
@WaltJr-OH This is presumably because XiangShan can only decode one vector instruction per cycle.

See: https://github.com/OpenXiangShan/XiangShan/issues/3488

WaltJr-OH, 2025-03-11:
> [@WaltJr-OH](https://github.com/WaltJr-OH) This is presumably because XiangShan can currently only decode one vector instruction per cycle.
> 
> See: [#4190](https://github.com/OpenXiangShan/XiangShan/issues/4190)

Thanks for responding, that explains the arithmetic peak data.

camel-cdr, 2025-03-22:
@WaltJr-OH The [lmul-1](https://github.com/OpenXiangShan/XiangShan/tree/lmul-1) branch, which will presumably get merged soon, modifies the decoder to support decoding vector instructions that only generate a single uop at the full decode width.
So it can decode six LMUL=1 vadd.vv per cycle and thus saturate the two execution units, but only decode a single LMUL=1 vadd.vx, because the .vx instructions generate an additional uop for the GPR -> VRF transfer.

rseac, 2025-05-27:
@Tang-Haojin I'm curious, is there a reason for the limitation of 128 bit for VLEN? Or is parameterization still in works ...