OpenXiangShan/XiangShan #4372: Regarding the development of configuration programs

Liujiaqi-jlu, 2025-03-06:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hello, I would like to ask some questions about configuring program development and workload. I want to compile workloads in an AM bare metal environment to implement some low-level functions, such as setting register values and enabling interaction between data and peripherals. But I am not very familiar with the bare metal environment of AM. I see that the KLIB library is included in the programs of the instances. If I write the configuration program in C language and compile it through AM, do I need to include some other libraries?

Tang-Haojin, 2025-03-06:
I believe if you only want to use AM to do some configuration, you do not have to include KLIB.

Liujiaqi-jlu, 2025-03-06:
> I believe if you only want to use AM to do some configuration, you do not have to include KLIB.

Thanks for your reply. I noticed that the external interfaces in the Chisel code include UART, flash, SD card, and interrupt interface. I want to achieve data exchange between the Xiangshan processor core and another computing chip unit, which should be achieved through peripheral interfaces. I feel that none of these peripheral interfaces are suitable. May I ask which interface should be used to achieve this.

Tang-Haojin, 2025-03-07:
> Thanks for your reply. I noticed that the external interfaces in the Chisel code include UART, flash, SD card, and interrupt interface. I want to achieve data exchange between the Xiangshan [...]

All these external interfaces in Chisel are for simulation purposes and are not real devices. For your question, UART can be used for data exchange between different devices.

closed 2025-09-05 by Yan-Muzi