OpenXiangShan/XiangShan #4385: `sltiu` instruction bug in NEMU

ha0lyu, 2025-03-10:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

[image]

[image]
NEMU throw `mcause=1`, and the `s9 = 0` not `1`.

### Expected behavior

NEMU execute this instruction correctly.

### To Reproduce

[testcase.zip](https://github.com/user-attachments/files/19157899/testcase.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: d6b0a27ff
- ready-to-run commit id: 8c943ff
- SPIKE commit id:

ha0lyu, 2025-03-10:
Thank you for your reply. Based on your explanation, I added the `sfence.vma` instruction after modifying `pmpcfg`. However, inconsistencies still occurred when executing this instruction:
[image]

[image]

I tested it on both NEMU and Spike. It can be seen that NEMU executed up to PC=0x800004ac, **yes NEMU execute `sltiu` correctly**, and, as expected, threw an illegal instruction exception:
[image]
NEMU commit id: 2235c04
[testcase.zip](https://github.com/user-attachments/files/19164772/testcase.zip)

Furthermore, I noticed in the first image that the **Commit Instr Trace** log shows `[12] commit pc 00000000800004a4`, followed by `[13] commit pc 0000000080000436`. This PC value doesn't appear to be the next sequential instruction; shouldn't it be `800004a8`?

NewPaulWalker, 2025-03-11:
> Based on your explanation, I added the `sfence.vma` instruction after modifying `pmpcfg`. However, inconsistencies still occurred when executing this instruction:

This is because the PC of your `sfence` instruction, after modifying the PMP, has already fallen within a region restricted by the PMP. When NEMU executes this instruction, it will naturally trigger an IAF (Instruction Access Fault).

ha0lyu, 2025-03-11:
Aha, I understand. NEMU does not wait for XiangShan to execute the `sfence` instruction. 
In this situation, where the test involves modifying PMP and requires XiangShan to flush its caches, how should we carry out such tests?

NewPaulWalker, 2025-03-11:
> Furthermore, I noticed in the first image that the **Commit Instr Trace** log shows `[12] commit pc 00000000800004a4`, followed by `[13] commit pc 0000000080000436`. This PC value doesn't appear to be the next sequential instruction; shouldn't it be `800004a8`?

In the log file, both the `Commit Group Trace` and `Commit Instr Trace` are circular buffers. The location pointed to by the arrow represents the content of the most recently committed instruction. Therefore, the 13th record is actually the earliest recorded entry, rather than the next instruction to be committed.  

You can analyze the `Commit Group Trace` and `Commit Instr Trace` together. Each entry in the `Commit Group Trace` corresponds to a ROB entry, where the `pc` field represents the PC of the first uop, and `cmtcnt` indicates the number of uops contained within that entry. The `Commit Instr Trace` only displays the first uop of each ROB entry because, in general cases, the first uop provides sufficient information. In special cases, a ROB entry typically contains only a single uop.

NewPaulWalker, 2025-03-11:
> Aha, I understand. NEMU does not wait for XiangShan to execute the `sfence` instruction.

Yes!

> In this situation, where the test involves modifying PMP and requires XiangShan to flush its caches, how should we carry out such tests?

Perhaps you can divide the code into two regions and apply different PMP protection strategies to them.

closed 2025-03-19 by ha0lyu