OpenXiangShan/XiangShan #4386: Xiangshan does not throw IAF trap

ha0lyu, 2025-03-10:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

An instruction at pc = 12870000 is not accessable:

[image]
Xiangshan executes it and gets `t0=1`:

[image]

[image]

### Expected behavior

Throw IAF trap.

### To Reproduce

[testcase.zip](https://github.com/user-attachments/files/19158117/testcase.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: d6b0a27ff
- ready-to-run commit id: 8c943ff
- SPIKE commit id:

ha0lyu, 2025-03-10:
NEMU does not throw IAF also.

NewPaulWalker, 2025-03-10:
Thank you for your issue. This is not a bug. Let me describe what happened:  

After executing an unknown instruction, the PC jumped to the address specified by `mtvec`, which is `0x12870000`. The address range `0x100000000-0x20000000` corresponds to the MMIO flash region. Although XiangShan's flash space is not that large, it matches only the lower bits. As a result, the value at `0x12870000` happens to fetch the first instruction in flash, which is `addiw t0, zero, 1`. After execution, the value of the `t0` register becomes `1`.  

In NEMU, however, `0x12870000` is an MMIO address but does not fall within the flash region. Accessing it triggers an IAF (Instruction Access Fault), but this is recorded in `mncause` instead of `mcause` (refer to the Sm/SsdbLtep extension for details). As a result, `mnepc` is set to `0x12870000`, meaning the addition instruction is not executed, and `t0` remains `0`.  

Regarding this behavioral difference, we currently have no plans to modify it.

ha0lyu, 2025-03-10:
Thanks for your reply. This discrepancy is affecting the testing process on XiangShan. It is suggested to consider prioritizing this issue in your upcoming tasks.

cebarobot, 2025-03-11:
> Thanks for your reply. This discrepancy is affecting the testing process on XiangShan. It is suggested to consider prioritizing this issue in your upcoming tasks.

I have to say, due to the limitation of the difftest framework and NEMU, we could not eliminate all Discrepancies between XiangShan and NEMU.

For this particular problem, we currently recommanded you to modify your testing framework to avoid this problem. We may to some refractor to fix this problem totally, but it takes time and is not the highest priority.

ha0lyu, 2025-03-11:
> Difftest serves as the foundation for ensuring the safety of XiangShan. 

Yes, we could not eliminate all discrepancies between XiangShan and NEMU, but we should strive to reduce these differences as much as possible in order to make XiangShan more secure and reliable.

I will adjust my testing procedures accordingly and look forward to seeing a perfect resolution to this issue.

closed 2025-09-05 by Yan-Muzi