OpenXiangShan/XiangShan #4388: Difference in the upper 16 bits of the PC being zero

ha0lyu, 2025-03-10:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

[image]
`s2 = 0xfffffffffffffbb5 `

[image]

The difference in the mepc and mtval values between NEMU and XiangShan lies in the upper 16 bits.

[image]

### Expected behavior

The values of these two registers should be consistent.

### To Reproduce

[testcase.zip](https://github.com/user-attachments/files/19158658/testcase.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: d6b0a27ff
- ready-to-run commit id: 8c943ff
- SPIKE commit id:

sinceforYy, 2025-03-11:
@ha0lyu 

Thanks for your question. we have reproduced the issue as described below. 

The value of register `s2` is `0xffff_ffff_ffff_fbb5`. `jr` is a pseudo instruction for the plain unconditional indirect jump instruction `jalr` with `rd=x0`. `jr s2` is a procedure return in the standard calling convention, where `rd=x0`, `rs1=s2` and `imm=0`.

> RISC-V spec for JALR:
> The target address is obtained by adding the sign-extended 12-bit I-immediate to the register rs1, then setting the least-significant bit of the result to zero.

Therefore, `s2 + imm = 0xffff_ffff_ffff_fbb5 + 0 = 0xffff_ffff_ffff_fbb5`, and set the least-significant bit of the result of zero, target address = `0xffff_ffff_ffff_fbb4`.

With no address translation/protection enabled and the physical address width of 48 bits,  this causes an instruction access fault(`IAF`) exception for the next instruction. Since the `IAF` for the next instruction is not generated normally, `mepc` and `mtval` are the result of high address truncation.

After fix this, the current test case enters an **infinite loop**. Is this the **expected behavior**?

ha0lyu, 2025-03-11:
Hi @sinceforYy, thanks for your detail explanation.
It could be an expected behavior if the CPU core is **repeatedly reset** within an infinite loop.

sinceforYy, 2025-03-11:
Ok, here is the PR that fixed it.  [fix(reidrectGen): fix redirectGen valid signal](https://github.com/OpenXiangShan/XiangShan/pull/4392)

closed 2025-03-12 by Tang-Haojin