OpenXiangShan/XiangShan #4401: NEMU has a problem when executing the `addi` instruction.

ha0lyu, 2025-03-11:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

There are two test cases, both of which use the `addi` instruction to update register values. NEMU, however, did not execute the addition operation correctly. 

Case A:
[image]
[image]
[image]

Case B:
[image]
[image]

### Expected behavior

The added value should be same.

### To Reproduce

[testcase.zip](https://github.com/user-attachments/files/19196263/testcase.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: d6b0a27
- NEMU commit id: 2235c04
- ready-to-run commit id: 8c943ff

### Additional context

I wonder if this could be related to NEMU's slower execution speed?

NewPaulWalker, 2025-03-12:
In case A, a self-modifying event occurred at address 0x800005bc. XiangShan still used the old instruction, where 0x80000522 + 196 = 0x800005e6, while NEMU used the new instruction after the self-modification, so `ra` was not updated. Why do you say that NEMU failed to execute correctly?

Case B is also a self-modifying program.

ha0lyu, 2025-03-12:
> Why do you say that NEMU failed to execute correctly?
Cause `ra` was not updated.

ha0lyu, 2025-03-12:
> Case B is also a self-modifying program.

Why do you think so?
In case B, the program load a value in data section, and modify it then store it.

NewPaulWalker, 2025-03-12:
> Why do you think so?
> In case B, the program load a value in data section, and modify it then store it.

Because I observed in NEMU that your test wrote 0 to address 0x80000118.

[image]

NewPaulWalker, 2025-03-12:
> Cause `ra` was not updated.

So have you tested whether Spike updates `ra` afterward?

ha0lyu, 2025-03-12:
Thank you very much for your help! I found that the issue was caused by a bug in our instruction test template, which led to this series of problems. I sincerely apologize for the trouble it caused you, and I’m truly grateful for your patience and support. Furthermore, I will fix the test template as soon as possible and proceed with further testing.

closed 2025-03-12 by ha0lyu

NewPaulWalker, 2025-03-12:
If other similar issues have no problem, please close it.