OpenXiangShan/XiangShan #4417: java.lang.OutOfMemoryError

wondersylar, 2025-03-13:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

我在make verilog时发现错误
报出java.lang.OutOfMemoryError
[image]
然后持续输出下面log
[image]

我已经参考了其他issue的建议，调整内存如下
export JAVA_OPTIONS="-XX:-UseGCOverheadLimit: -Xms256m -Xmx64g -Xss256m"
我在程序运行时查看可用内存，还有25G可用
[image]

1. 我发现已经编译出/xs_env/XiangShan/build/rtl/XSTop.fir，是否可以单独执行下面命令，运行下面命令需要多久编译出verilog？
firtool /home/ubuntu01/xiangshan_proj/xs_env/XiangShan/build/rtl/XSTop.fir -warn-on-unprocessed-annotations -output-annot
tion-file circt.anno.json --repl-seq-mem --repl-seq-mem-ffile=XSTop.sv.conf -0=release - disable-annotation-unknown - lowering-options=explicitBitcast,disallowLocalVariables,disallowPortDecl
haring,locationInfoStyle=none - - ignore-read-enable-mem -verbose-pass-executions -mlir-print-ir-after-all --split-verilog -o=build/rtl
2. 如何进一步debug，还可以调整哪些参数？
3. 是否有其他工程可以进行编译测试

### What did you do before

我已经参考了其他issue的建议，调整内存如下
export JAVA_OPTIONS="-XX:-UseGCOverheadLimit: -Xms256m -Xmx64g -Xss256m"
我在程序运行时查看可用内存，还有25G可用
[image]

### Environment

- XiangShan branch: master
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: ubuntu 24.04
- gcc version: gcc (Übuntu 13.3.0-6ubuntu2~24.04) 13.3.0
- mill version: Mill Build Tool version 0.11.8
- java version: Java version: 11.0.26,   openjdk 11.0.26 2025-01-21

Ma-YX, 2025-03-14:
It seems that your total Mem+Swap is 79 GB, you can try to expand Swap or adjust Mem `export JAVA_OPTIONS="-XX:-UseGCOverheadLimit: -Xms256m -Xmx32g -Xss256m"` (make it less until compilation works)

Ma-YX, 2025-03-14:
As for other project, you can try different version of XiangShan. E.g. last generation of XiangShan --- Nanhu

wondersylar, 2025-03-24:
谢谢 我切换了master分支，可以成功编译了

closed 2025-03-24 by wondersylar