OpenXiangShan/XiangShan #4428: `isa pmp check failed` when running the XiangShan simulation example in README

JacyCui, 2025-03-17:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

I was running XiangShan (v2.1) simulation according to README.

First, I run the following command to build the default simulator.

```shell
~/XiangShan$ time make emu
... ...
real    392m3.772s   #  6.5 h
user    693m43.122s  # 11.5 h
sys     31m31.151s   # 31.5 m
```

Then, I wanna try to see if the `./build/emu` work well, and run the following command provided by README.

```shell
~/XiangShan$ ./build/emu -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so
```

But got:
```
...
[src/memory/paddr.c:171,paddr_read] isa pmp check failed
ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
...
```

### What did you do before

I searched through issues and documents and thought maybe I should update `ready-to-run` submodule to the latest version, so I

```shell
cd ready-to-run
git pull
git checkout master
```

Then, I type the same instruction written in README:
```
~/XiangShan$ ./build/emu -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so
```

But still got the similar error:

```
~/XiangShan$ ./build/emu -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so
Emu compiled at Mar 16 2025, 19:38:05
Using simulated 32768B flash
[warning]no valid flash bin path, use preset flash instead
The image is ./ready-to-run/coremark-2-iteration.bin
Using simulated 8192MB RAM
NemuProxy using ./ready-to-run/riscv64-nemu-interpreter-so
The first instruction of core 0 has commited. Difftest enabled. 

============== Commit Group Trace (Core 0) ==============
commit group [00]: pc 1ffff80000 cmtcnt 1 <--
commit group [01]: pc 0000000000 cmtcnt 0
commit group [02]: pc 0000000000 cmtcnt 0
commit group [03]: pc 0000000000 cmtcnt 0
commit group [04]: pc 0000000000 cmtcnt 0
commit group [05]: pc 0000000000 cmtcnt 0
commit group [06]: pc 0000000000 cmtcnt 0
commit group [07]: pc 0000000000 cmtcnt 0
[... 137 lines cut ...]
### Environment

- XiangShan branch: `master`
- XiangShan commit id: `f0d8a1cf543c113433ff98140b6d77055d930deb` (`v2.1`)
- NEMU commit id: `c1ae161da33c3b32ef6b79814b48df4adef68b5c` (`master`)
- Operating System: `Ubuntu 22.04`
- gcc version: `gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0`
- mill version: `Mill Build Tool version 0.9.8`
- java version: `openjdk 17.0.2 2022-01-18`

### Additional context

I have directly used

```shell
make emu
```

instead of 

```shell
make emu CONFIG=MinimalConfig EMU_THREADS=2 -j10
```

Do I have to use `MinimalConfig` to make the simulator work?

cebarobot, 2025-03-17:
Thanks for your issue.

Please use the `ready-to-run` provided by `XiangShan` repo, as it includes the appropriate prebuilt NEMU that is compatible with the XiangShan. No need to switch master.

The NEMU and difftest API changes a lot. The master NEMU is not compatible with the Southlake version (v2.1) of XiangShan.

By the way, whether or not you select MinmalConfig has no impact on this diff error.

JacyCui, 2025-03-17:
Thanks for your time :)

At the very beginning, I did use the `ready-to-run` provided by `XiangShan` repo, whose commit id is `4ba2822db1cfee003780193107741418763b3eed` .

As I described above (in the **Describe you problem** section), I ran

```shell
~/XiangShan$ ./build/emu -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so
```

but got

```
...
[src/memory/paddr.c:171,paddr_read] isa pmp check failed
ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
...
```

I did noticed an issue OpenXiangShan/XiangShan#2886 encountered with the similar error message but didn't find enough information to guide me out of the trouble.

The README file guided me to set `NEMU_ROOT` and `AM_HOME` environment variables before I ran `make emu`. I set them to the correponding latest repositories' root (i.e. simply `git clone` `NEMU` and `AM`) since I didn't know the proper commit id of them that I should use. Does this matter?

NewPaulWalker, 2025-03-19:
Thank you for your issue！

> At the very beginning, I did use the `ready-to-run`, whose commit id is `4ba2822db1cfee003780193107741418763b3eed`, provided by `XiangShan` repo.
> 
> As I described above (in the **Describe you problem** section), I ran
> 
> ~/XiangShan$ ./build/emu -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so
> but got
> 
> ```
> ...
> [src/memory/paddr.c:171,paddr_read] isa pmp check failed
> ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
> ...
> ```

Yes, you are right！ I compiled an emu(XiangShan version 2.1) and used the ready-to-run provided by the XiangShan repository with commit ID 4ba2822db1cfee003780193107741418763b3eed. After running the workload, I encountered a series of 
```
ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
```

Then, I tried compiling a `riscv64-nemu-interpreter-so` from NEMU's southlake branch: [https://github.com/OpenXiangShan/NEMU/commits/southlake/](https://github.com/OpenXiangShan/NEMU/commits/southlake/). When I ran the simulation program afterward, I found that NEMU triggered an illegal instruction on an `sd` instruction, which is clearly incorrect.
You can check my simulation log:
[emu.log](https://github.com/user-attachments/files/19335525/emu.log)

I found that this issue has already been fixed in NEMU's master branch. Moreover, XiangShan (v2.1) may not be fully compatible with the `riscv64-nemu-interpreter-so` and `coremark-2-iteration.bin` from the ready-to-run. Additionally, these components are already several years old. **I strongly recommend that you try testing with the latest master branch of XiangShan.**

> The README file guided me to set `NEMU_ROOT` and `AM_HOME` environment variables before I ran `make emu`. I set them to the correponding latest repositories' root (i.e. simply `git clone` `NEMU` and `AM`) since I didn't know the proper commit id of them that I should use. Does this matter to the error I ran into?

This is unrelated to the issue you're facing.

JacyCui, 2025-03-21:
I understand your recommendation to use the latest master branch. However, I have chosen the Southlake version for my research due to its status as the latest stable release, the most recently taped-out version, and its use of the Scala Firrtl Compiler (SFC) instead of Firtool (provided by CIRCT).

As part of my research, I am investigating the time required for Southlake to complete a full cycle of comprehensive system-level simulation-based verification. I don’t have to know the exact time—an approximation (e.g., several seconds, minutes, hours, or days) would be sufficient and greatly appreciated.

Thank you in advance for your assistance!

closed 2025-04-07 by JacyCui