OpenXiangShan/XiangShan #4433: About customizing Xiangshan configuration

Liujiaqi-jlu, 2025-03-18:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hello, while reading the Soc Introduction module in the Xiangshan documentation, I noticed that we can specify different architecture parameters. I would like to ask, can the address range of AXI's three interfaces be customized, and where do I need to customize it? How to handle the update of default PMA configuration mentioned in the document? Thank you！

Ma-YX, 2025-03-19:
Thanks for your issue.

Plz provide the generation/version of XiangShan that you want to ask. You mentioned AXI, so is it nanhu(here https://docs.xiangshan.cc/zh-cn/latest/integration/soc/ is nanhu as well)? While kunminghu uses CHI now, there are some differences.

Liujiaqi-jlu, 2025-03-19:
> Thanks for your issue.
> 
> Plz provide the generation/version of XiangShan that you want to ask. You mentioned AXI, so is it nanhu(here https://docs.xiangshan.cc/zh-cn/latest/integration/soc/ is nanhu as well)? While kunminghu uses CHI now, there are some differences.

Yes, I am currently using the Nanhu version and the Docker image environment.

Ma-YX, 2025-03-21:
To change default PMA configuration, you can go to PMA.scala in XiangShan's backend folder.

As for AXI's three interfaces, we do not recommend you to change them, since changes of them are not fully verified. However, you can use an AXI Bridge to solve the problem.

Liujiaqi-jlu, 2025-03-23:
> To change default PMA configuration, you can go to PMA.scala in XiangShan's backend folder.
> 
> As for AXI's three interfaces, we do not recommend you to change them, since changes of them are not fully verified. However, you can use an AXI Bridge to solve the problem.

OK，thanks

closed 2025-09-05 by Yan-Muzi