OpenXiangShan/XiangShan #4457: 如何生成chi port？

wondersylar, 2025-03-24:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

请问如何生成chi port？我目前在master分支上
make verilog CONFIG=KunminghuV2config NUM_CORES=2， XSTOP.sv 内是axi port

[image]
另外 请教下如何看香山有哪些可配参数，如何配置这些参数，感谢

### What did you do before

make verilog CONFIG=KunminghuV2config 和make emu CONFIG=KunminghuV2config 看起来都没有生成chi port

### Environment

- XiangShan branch: master
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System:  ubuntu 24.04
- gcc version: gcc (Übuntu 13.3.0-6ubuntu2~24.04) 13.3.0
- mill version: 0.12.3
- java version: Java version: 11.0.26, openjdk 11.0.26 2025-01-21

Ma-YX, 2025-03-25:
Thanks for your issue.

Currently, CHI protocol is only used between CoulpledL2 and OpenLLC/OpenNCB (L2 $ - L3 $), under CONFIG=KunminghuV2config. 

Ports in XSTOP.sv is to memory/device, XiangShan uses AXI4 here.

Tang-Haojin, 2025-03-25:
You can try `make verilog CONFIG=XCNoCTopConfig`. This will generate CHI ports with no L3 cache connected.

wondersylar, 2025-03-26:
Thank you for your reply. I would like to further ask if all configurations can be viewed from config.scala, and is there any documentation that describes the configurable aspects of XiangShan?

wondersylar, 2025-03-27:
> You can try `make verilog CONFIG=XCNoCTopConfig`. This will generate CHI ports with no L3 cache connected.

Tang-Haojin, 2025-03-27:
Yes, all the validated configurations can be found in config.scala. For configurable parameters, we have an incomplete list (available in Chinese only). Hope this can help you.

[香山CPU配置项目.xlsx](https://github.com/user-attachments/files/19480568/CPU.xlsx)

wondersylar, 2025-03-27:
thank you for your share
If  configure the project by make verilog CONFIG=KunminghuV2config NUM_CORES=2,  the project will includes L3 and has two cores?
If  configure the project by  make verilog CONFIG=XCNoCTopConfig, the project will exclude L3, and also not allow configuring it with two cores? 
Is this understanding correct?

Tang-Haojin, 2025-03-27:
L3 in KunminghuV2Config is called OpenLLC.

closed 2025-09-05 by Yan-Muzi