OpenXiangShan/XiangShan #4459: Regarding the version issue of Vivado

Liujiaqi-jlu, 2025-03-25:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hello, I have compiled the Xiangshan kernel for the Nanhu version according to the FPGAprototype guide. I notice that we will need to use Vivado in the future. What are the requirements for the version of Vivado? thank you!

xianfeili, 2025-03-27:
The project we provide is base on VIvado 2020.2

Liujiaqi-jlu, 2025-03-27:
> The project we provide is base on VIvado 2020.2

ok，thanks

closed 2025-03-29 by Tang-Haojin