OpenXiangShan/XiangShan #4467: Regarding the issue of mismatched FPGA simulation interface names

Liujiaqi-jlu, 2025-03-27:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

Hello, when I followed the instructions in the first picture to perform the Vivado simulation to the step of "make update_comre_flist CORE-DIR=$NOOP-HOME/build", I encountered the following error message, which shows that the mem port does not match. As shown in the second and third images, in the instantiated u_stxTOP, there is a '0' in the middle of the corresponding mem port. But in xstop.sv, there is no '0' in the middle of the ports related to mem. Although it is possible to continue running temporarily by changing the name, may I ask if this is a configuration issue or some other bug. thank you!

[image]

[image]

[image]

### Expected behavior

It should be carried out normally according to the simulation process.

### To Reproduce

Follow the steps of nanhu in FPGAPrototype completely, and only make some modifications to the generated code in the third step, which has not been modified yet.

### Environment

- XiangShan branch: 09d2a32142c64fca898e17c0b943e61ddc331958
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id:

xianfeili, 2025-03-28:
I follow the command describe in Xiangshan docs to generate the Xiangshan RTL.  

```
git clone https://github.com/OpenXiangShan/XiangShan
cd XiangShan
export NOOP_HOME=$(pwd)
git checkout 09d2a32142c64fca898e17c0b943e61ddc331958
make init
make clean
make verilog NUM_CORES=2 # Dual-Core XiangShan Nanhu
```

The module XSTop in XSTop.v file have the "0" in the middle of the AXI port.

```
module XSTop(
  output         dma_0_awready,
  input          dma_0_awvalid,
  input  [13:0]  dma_0_awid,
  input  [35:0]  dma_0_awaddr,
  input  [7:0]   dma_0_awlen,
  input  [2:0]   dma_0_awsize,
  input  [1:0]   dma_0_awburst,
  input          dma_0_awlock,
  input  [3:0]   dma_0_awcache,
  input  [2:0]   dma_0_awprot,
  input  [3:0]   dma_0_awqos,
  output         dma_0_wready,
  input          dma_0_wvalid,
  input  [255:0] dma_0_wdata,
  input  [31:0]  dma_0_wstrb,
  input          dma_0_wlast,
  input          dma_0_bready,
  output         dma_0_bvalid,
  output [13:0]  dma_0_bid,
  output [1:0]   dma_0_bresp,
  output         dma_0_arready,
  input          dma_0_arvalid,
  input  [13:0]  dma_0_arid,
  input  [35:0]  dma_0_araddr,
  input  [7:0]   dma_0_arlen,
  input  [2:0]   dma_0_arsize,
  input  [1:0]   dma_0_arburst,
  input          dma_0_arlock,
  input  [3:0]   dma_0_arcache,
  input  [2:0]   dma_0_arprot,
  input  [3:0]   dma_0_arqos,
  input          dma_0_rready,
  output         dma_0_rvalid,
  output [13:0]  dma_0_rid,
  output [255:0] dma_0_rdata,
  output [1:0]   dma_0_rresp,
  output         dma_0_rlast,
```

So, I think it must be the generate env. issue. Maybe you can provide the your env. setting. So we can locate the root cause.

Liujiaqi-jlu, 2025-04-09:
> I follow the command describe in Xiangshan docs to generate the Xiangshan RTL.
>
> ```
> git clone https://github.com/OpenXiangShan/XiangShan
> cd XiangShan
> export NOOP_HOME=$(pwd)
> git checkout [...]

Thanks for your reply. Due to some issues with the school server network, we are unable to access GitHub and Mill, which has temporarily prevented us from reproducing and generating the corresponding code. May I ask if you can provide the RTL code of the FPGA version generated according to the above steps? Thank you very much!

[image]

xianfeili, 2025-04-10:
[build_tmp.tar.gz](https://github.com/user-attachments/files/19682915/build_tmp.tar.gz) You can try this attach.

Liujiaqi-jlu, 2025-04-11:
> [build_tmp.tar.gz](https://github.com/user-attachments/files/19682915/build_tmp.tar.gz) You can try this attach.

Thank you for your reply. I noticed that there is a significant difference between the code you generated and my previous version. The version of RTL code I generated earlier had a lot of. sv files, and the final xstop was also in. sv format, but only 2000 lines. I think the version you generated should have consolidated all the code into the xstop file. May I ask if you can send me the entire build folder you generated? I noticed that the build folder I previously generated had some configuration files besides RTL. I want to generate a Vivado project, which requires referencing the build folder.

xianfeili, 2025-04-11:
Build folder is too large.

Liujiaqi-jlu, 2025-04-11:
> Build folder is too large.

Okay, I tried again. I created the project according to 'env scripts', and when instantiating XSTop, I used port names without' 0 '. However, the version we generated and the one you provided now both have' 0 ', which does not match the port instantiated in the FPGA project. There are some issues here, and I am not sure if it is a problem with the Vivado project script. If possible, could you package a complete Vivado project for us to run. Thank you.

xianfeili, 2025-04-14:
The XSTop_wrapper.v is the file you use in the project? https://github.com/OpenXiangShan/env-scripts/blob/main/xs_nanhu_fpga/src/rtl/XSTop_wrapper.v

You make the vivado project use the `make nanhu` right ?

closed 2025-09-10 by Yan-Muzi