OpenXiangShan/XiangShan #4492: make verilog report no config file

huilongs999, 2025-04-02:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

make verilog mkdir -p build/rtl time -avp -o ./build/time.log mill -i xiangshan.runMain top.TopMain   \         --target-dir build/rtl --config DefaultConfig --issue B --firtool-opt "--repl-seq-mem --repl-seq-mem-file=XSTop.sv.conf"                \         --num-cores 1 --dump-fir --target systemverilog --split-verilog --firtool-opt "-O=release --disable-annotation-unknown --lowering-options=explicitBitcast,disallowLocalVariables,disallowPortDeclSharing,locationInfoStyle=none" --fpga-platform --disable-all --remove-assert --reset-gen --disable-always-basic-diff ./scripts/gen_sep_mem.sh "./scripts/vlsi_mem_gen" "build/rtl/XSTop.sv.conf" "build/rtl" 
cat: build/rtl/XSTop.sv.conf: No such file or directory 
rm: cannot remove 'build/rtl/XSTop.sv.conf.tmp': No such file or directory 
make: *** [Makefile:155: build/rtl/XSTop.sv] Error 1

### What did you do before

我的环境是ubuntu20.4,按照环境搭建要求，依次执行了如下命令
1.git clone https://github.com/OpenXiangShan/xs-env.git
2.cd xs-env  sudo -s ./setup-tools.sh
3. ./setup.sh
4. source env.sh
5. 在/xs-env/XiangShan 下执行 make init
6. 在 /xs-env/XiangShan 下运行 make verilog，报错。
重复执行上述step1~6，还是报错。

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Tang-Haojin, 2025-04-02:
Please provide the full log by `make verilog >log.txt 2>err.txt` and upload log.txt and err.txt to this issue. Thanks.

huilongs999, 2025-04-02:
[log.txt](https://github.com/u

[err.txt](https://github.com/user-attachments/files/19562054/err.txt)

ser-attachments/files/19562046/log.txt)

huilongs999, 2025-04-02:
the  log file context is:
mkdir -p build/rtl
time -avp -o ./build/time.log mill -i xiangshan.runMain top.TopMain   \
	--target-dir build/rtl --config DefaultConfig --issue B --firtool-opt "--repl-seq-mem --repl-seq-mem-file=XSTop.sv.conf"		\
	--num-cores 1 --dump-fir --target systemverilog --split-verilog --firtool-opt "-O=release --disable-annotation-unknown --lowering-options=explicitBitcast,disallowLocalVariables,disallowPortDeclSharing,locationInfoStyle=none" --fpga-platform --disable-all --remove-assert --reset-gen --disable-always-basic-diff
./scripts/gen_sep_mem.sh "./scripts/vlsi_mem_gen" "build/rtl/XSTop.sv.conf" "build/rtl"

the err file context is ：
cat: build/rtl/XSTop.sv.conf: No such file or directory
rm: cannot remove 'build/rtl/XSTop.sv.conf.tmp': No such file or directory
make: *** [Makefile:155: build/rtl/XSTop.sv] Error 1

Tang-Haojin, 2025-04-02:
That REALLY looks strange. Can you give me a full screenshot of the output of `make verilog`?

closed 2025-09-10 by Yan-Muzi