OpenXiangShan/XiangShan #4499: Error when doing sim

rseac, 2025-04-03:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

When compiling XiangShan with `make sim-verilog`, I get the following error.

```
[768] [warn] There were 148 warning(s) during hardware elaboration.
[768/768] =============== xiangshan.test.runMain top.SimTop --target-dir build/rtl --confi...ring,locationInfoStyle=none --enable-difftest --full-stacktrace =============== 520s
1 tasks failed
xiangshan.test.runMain Subprocess failed
make: *** [Makefile:229: build/rtl/SimTop.sv] Error
```

### What did you do before

I am trying to setup XiangShan.

```bash
git clone https://github.com/OpenXiangShan/xs-env.git
./setup-tools.sh
source setup.sh
source update-submodule.sh
source env.sh
cd /xs-env/XiangShan
make init
make sim-verilog
```

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Tang-Haojin, 2025-04-03:
Seems that it is caused by insufficient memory. How many RAM do you have in your environment?

rseac, 2025-04-03:
`/proc/meminfo` shows this as the total: MemTotal:       32804376 kB

Tang-Haojin, 2025-04-03:
That may be the reason. I have just updated xs-env. You can try to clone it again and run what you have done one more time. But this time, you may append `JVM_XMX=20G` after `make sim-verilog`. This can reduce the memory consumption.

rseac, 2025-04-16:
This worked! Thank you.  I'll close the issue.

closed 2025-04-16 by rseac