OpenXiangShan/XiangShan #4504: ICacheMissEntry Assertion Violable

JacyCui, 2025-04-04:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

I'm working with the latest stable release, Southlake version, of XiangShan.
And I found an assertion violation when exercising its `ICacheMissEntry` module.
I'm wondering whether it can be considered a bug.
The corresponding optimized low firrtl file is as follows, which is reffered to as `ICacheMissEntry.opt.lo.fir` later.
```firrtl
FIRRTL version 1.1.0
circuit ICacheMissEntry :
  module ICacheMissEntry :
    input clock : Clock
    input reset : UInt<1>
    input io_id : UInt<1>
    output io_req_ready : UInt<1>
    input io_req_valid : UInt<1>
    input io_req_bits_paddr : UInt<38>
    input io_req_bits_vaddr : UInt<39>
    input io_req_bits_waymask : UInt<4>
    input io_req_bits_coh_state : UInt<2>
    output io_resp_valid : UInt<1>
    output io_resp_bits_data : UInt<512>
    output io_resp_bits_corrupt : UInt<1>
    input io_mem_acquire_ready : UInt<1>
    output io_mem_acquire_valid : UInt<1>
    output io_mem_acquire_bits_opcode : UInt<3>
    output io_mem_acquire_bits_param : UInt<3>
    output io_mem_acquire_bits_size : UInt<3>
    output io_mem_acquire_bits_source : UInt<3>
    output io_mem_acquire_bits_address : UInt<38>
    output io_mem_acquire_bits_user_alias : UInt<2>
    output io_mem_acquire_bits_user_preferCache : UInt<1>
    output io_mem_acquire_bits_user_needHint : UInt<1>
    output io_mem_acquire_bits_echo_blockisdirty : UInt<1>
    output io_mem_acquire_bits_mask : UInt<32>
    output io_mem_acquire_bits_data : UInt<256>
    output io_mem_acquire_bits_corrupt : UInt<1>
    output io_mem_grant_ready : UInt<1>
    input io_mem_grant_valid : UInt<1>
    input io_mem_grant_bits_opcode : UInt<3>
    input io_mem_grant_bits_param : UInt<2>
    input io_mem_grant_bits_size : UInt<3>
    input io_mem_grant_bits_source : UInt<3>
    input io_mem_grant_bits_sink : UInt<6>
    input io_mem_grant_bits_denied : UInt<1>
    input io_mem_grant_bits_echo_blockisdirty : UInt<1>
    input io_mem_grant_bits_data : UInt<256>
    input io_mem_grant_bits_corrupt : UInt<1>
    input io_mem_finish_ready : UInt<1>
    output io_mem_finish_valid : UInt<1>
    output io_mem_finish_bits_sink : UInt<6>
    input io_meta_write_ready : UInt<1>
    output io_meta_write_valid : UInt<1>
    output io_meta_write_bits_virIdx : UInt<8>
    output io_meta_write_bits_phyTag : UInt<26>
    output io_meta_write_bits_coh_state : UInt<2>
    output io_meta_write_bits_waymask : UInt<4>
    output io_meta_write_bits_bankIdx : UInt<1>
    input io_data_write_ready : UInt<1>
    output io_data_write_valid : UInt<1>
    output io_data_write_bits_virIdx : UInt<8>
    output io_data_write_bits_data : UInt<512>
    output io_data_write_bits_waymask : UInt<4>
    output io_data_write_bits_bankIdx : UInt<1>
[... 325 lines cut ...]
poke io_mem_grant_bits_size 6
step
```

To reproduce this violation, run the following instruction:

```shell
$ treadle -tfsf ICacheMissEntry.opt.lo.fir < refill_done.payload
```

And the violation is indicated by simulator outputs:

```
...
Assertion failed: refill not done!
    at ICacheMissUnit.scala:171 assert(refill_done, "refill not done!")
Error: exception Failure Stop:assert:(65) at  @[ICacheMissUnit.scala 171:19] treadle.executable.StopException: Failure Stop:assert:(65) at  @[ICacheMissUnit.scala 171:19]
...
```

Here, `treadle` is the [firrtl simulator](https://github.com/chipsalliance/treadle/blob/v1.5.6/utils/bin/treadle), whose version is `v1.5.6`.

### Environment

- XiangShan commit id: `v2.1` tag

Ma-YX, 2025-04-07:
Thanks for your question.

Actually, this is not a bug. Not all inputs are legal for certain hardware design. We use assertions as constraint condition here to avoid  illegal inputs and help locating problems.

The situation your encountered is legal in verilog.

JacyCui, 2025-04-07:
May I ask under what circumstances an assertion violation should be considered as a bug and under what circumstances it shouldn't ?

eastonman, 2025-04-07:
@ngc7331 are you familar with V2.1 code? Is this stimulation legal?

ngc7331, 2025-04-07:
> [@ngc7331](https://github.com/ngc7331) are you familar with V2.1 code? Is this stimulation legal?

I'm not very familiar with it, but I remember it using the TL-C protocol, so I guess `poke io_mem_grant_bits_opcode 1` (`AccessAckData` in TL-UH/UL) is not correct, it should be 5(`Grant`)?

Also I'm not sure whether `poke io_mem_grant_bits_size 6` after `poke io_mem_grant_valid 1` and `step` is correct.

> May I ask under what circumstances an assertion violation should be considered as a bug and under what circumstances it shouldn't ?

It's hard to tell, the assertion in this issue is checking the behavior of TileLink bus (or, L2 cache), not ICacheMissUnit itself, so its failure should be considered a test stimulation error (in UT) or L2 cache bug (in IT/ST) rather than an ICacheMissUnit bug. Other assertions may be the opposite, with their failure indicating a bug. You may need to refer to the scala code to determine.

JacyCui, 2025-04-09:
Thanks so much for your feedback!

If I understand correctly, there isn’t currently a general automated way to determine whether an assertion violation is truly a bug — because it’s hard to tell whether the violation comes from a flaw in the design itself or from incorrect usage. Is that right?

I’m from the programming languages (PL) community, and my research focuses on program analysis, which has been shown to be very effective for automated bug detection in software. Since HDLs are a specific kind of programming language, I believe there’s a lot of untapped potential in applying similar analysis techniques. I chose Chisel as my entry point because of its vibrant, open, and diverse ecosystem, which contrasts with the relatively closed nature of Verilog (I love open-source :) ). Moreover, Chisel is emerging as a leading modern HDL. 

While working on this, one thing really puzzled me: **I did a broad survey and found that HDL programs don’t seem to exhibit the kinds of general bugs we commonly see in software — things like use-before-initialization, null pointer dereference, array index out of bounds, unsafe type casts, division by zero, memory leaks, etc.**

That led me to focus on assertion violations as a potential source of bugs. I built a lightweight analysis that found 28 assertion violations in the latest version of XiangShan (millions of lines of code) in under a minute. I manually wrote test payloads to trigger 22 of them — the rest were too complex to handle by hand. But here’s the frustrating part: even when I can show that the violation is reproducible, it doesn’t necessarily mean it’s a bug. That ambiguity makes the whole process quite challenging.

Still, I think automated bug detection is very promising for hardware. From what I’ve seen, hardware projects often spend more time finding bugs than actually developing features.

So I’d really appreciate insights from those of you with experience in open-source hardware development:
**Are there common or recurring types of bugs in hardware designs, similar to the general bugs we see in software?**

Looking forward to learning from your experiences!

Ma-YX, 2025-04-09:
That's a good question. We welcome you who view hardware design / programming language from software side.

Your understaning is correct. When we design hardware components, we should follow corresponding rules, like RISC-V manul, CHI protocol. Therefore, certain combinations of signal(value) are illeagel, which we use assertions to avoid or locate. To conduct your research on certain hardware project, you may refer to relative SPEC at first, then using contraint inputs for automated bug detection.

As for common or recurring types of bugs in hardware designs, there does exist some. For example, `use of register before initialization`, which will cause the spread of `X` value, and `Combinational Logic Loop`, etc.

JacyCui, 2025-04-10:
I understand that certain combination of signal values are illegal. For example, when using a RISC-V SoC, I should input something confined to RISC-V specification. But what if I'm an attacker and I maliciously input something that is out of the scope of specification? Will this be handled properly?

Ma-YX, 2025-04-10:
Actually, the assertions that we used here are for simulation verification. As for real chip, those assertions will be deleted (after synthesis).

Attackers are not supposed to have the access to manipulate input of component of real chip without physical contact. Even if attack is launched from peripherals, peripheral controllers (like USB controller) should filter illegal input.