OpenXiangShan/XiangShan #4505: dispatchData.io

mlabaf2, 2025-04-06:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hi
What is the source of dispatchData.io inputs and destination of dispatchData.io outputs in Rob.scala? 
I have some extra outputs from Alu.scala which I want to save them in register file in addition to just one data output of io.out.bits.data in Alu.scala destination . What is the path to send them to register file? I defined them in RobCommitInfo and also in ExuInput and ExuOutput.....
Thannks

lewislzh, 2025-04-07:
The current Nanhu version code is organized as follows: versions prior to nanhu-v2 are stored in this repository, while nanhu-v3/v3a are located in the personal repository of [siudya](https://github.com/Siudya/Nanhu). 
Versions nanhu-v2/v3/v3a have all undergone tape-out validation and contain known issues that have been resolved in the nanhu-v5 repository. 
If you are using Nanhu due to the relatively large scale of Kunminghu, it is recommended to use the code from the nanhu-v5 repository. We have communicated with the Nanhu team, and they will respond to issues in the v5 repository. If no response is provided, please feel free to open an issue in this repository to remind us.

mlabaf2, 2025-04-09:
please answer my question in https://github.com/OpenXiangShan-Nanhu/Nanhu-V5/issues
thanks

mlabaf2, 2025-04-10:
> In the Nanhu version of the ROB, the `dispatchData` you mentioned is mainly used to store uop information when instructions are enqueued in ROB after the dispatch stage. Its index can be understood as the index of the ROB entry. This information is used during commit/walk stages for purposes like difftest comparison and sending updates to the frontend FTQ.
> 
> As for your intention to add additional read/write inputs in the ALU, I believe it doesn't involve modifications to `robCommitInfo`. The ROB's commit information is not directly related to the register file writeback. Fields like `ldest` are primarily used for debugging or difftest purposes. You may need to focus more on the implementation details of the writeback arbiter. For the specific implementation of the functional unit writeback to the regfile network in Nanhu, it's recommended to raise the question in the [`nanhu-v5`](https://github.com/OpenXiangShan-Nanhu/Nanhu-V5) repository

> In the Nanhu version of the ROB, the `dispatchData` you mentioned is mainly used to store uop information when instructions are enqueued in ROB after the dispatch stage. Its index can be understood as the index of the ROB entry. This information is used during commit/walk stages for purposes like difftest comparison and sending updates to the frontend FTQ.
> 
> As for your intention to add additional read/write inputs in the ALU, I believe it doesn't involve modifications to `robCommitInfo`. The ROB's commit information is not directly related to the register file writeback. Fields like `ldest` are primarily used for debugging or difftest purposes. You may need to focus more on the implementation details of the writeback arbiter. For the specific implementation of the functional unit writeback to the regfile network in Nanhu, it's recommended to raise the question in the [`nanhu-v5`](https://github.com/OpenXiangShan-Nanhu/Nanhu-V5) repository.

Hi
Thanks for response.
In addition to Data in/out, I have addition control signal from exe to other modules like regfile or csr ....
How can I send these signals from exe to them? is it by robCommitInfo?

I am using f580a020c2c5a94799a1df45c3e120f37acc2f83 which different from Nanhu link which you sent....

lewislzh, 2025-04-10:
The version you are currently using seems to be from an intermediate development commit between NanHu and Kunminghu, and I cannot be certain which exact version you have adopted. Regarding your question, I can only explain the current implementation status on the Kunminghu main branch.

To pass control signals to CSR, you can refer to the handling method of `fflags`: add your control signal to the functional unit's writeback bundle, store it into the `ROBEntry` when writing back to ROB, then retrieve it from the `ROBEntry` (commitInfo) at the commit stage, and pass it through the `CSRIO` path to the CSR module.

For passing control signals to the register file (Regfile), Kunminghu currently does not have such an implementation. Perhaps you can achieve this by introducing additional control signals in the writeback path and handling them appropriately in the Regfile.

lewislzh, 2025-04-10:
The current version status of XiangShan is as follows:
The first stable micro-architecture of XiangShan is called Yanqihu (雁栖湖) [on the yanqihu branch(Stable, no longer maintained)](https://github.com/OpenXiangShan/XiangShan/tree/yanqihu), which has been developed since June 2020.

The second micro-architecture of XiangShan is called Nanhu (南湖).  
Multiple versions of Nanhu have been released or are under development:

- [Nanhu V2 (Stable, no longer maintained)](https://github.com/OpenXiangShan/XiangShan/tree/nanhu)  
- [Nanhu V3 (Stable, no longer maintained)](https://github.com/Siudya/Nanhu/tree/main)  
- [Nanhu V5 (Under ACTIVE Development)](https://github.com/OpenXiangShan-Nanhu/Nanhu-V5/tree/main) – this repository

The third version of XiangShan, also known as Kunminghu (昆明湖) [on the master branch (Under ACTIVE Development)](https://github.com/OpenXiangShan/XiangShan/tree/master).

Currently, this repository can only provide support for modifications based on the main branch. Issues arising from diverged branches may not be addressed effectively. It is recommended to use Nanhu-v5 (smaller scale) or the Kunminghu main branch for development.

closed 2025-09-10 by Yan-Muzi