OpenXiangShan/XiangShan #4536: make verilog for FPGA report could not find file

1024493603, 2025-04-09:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

我准备去编译香山昆明湖RTL烧录到我的FPGA板子上。
当我去执行文档中的`make kmh CORE_DIR=$NOOP_HOME/build`命令时报了如下的错误。
[image]
未截图到的部分为`xs_kmh_fpga/src/tcl/xs_uart.tcl`文件的内容。
根据执行的命令`vivado -mode batch -source src/tcl/xs_uart.tcl -tclargs --cpu kmh --project_name xs_kmh --core_dir /xs-env/XiangShan/build --ip_dir "/nfs/home/share/fpga/vivado/XS_100NL_ST_V3_TAG_20211229"`来看猜测是ip_dir的问题，但我不清楚这里应该指向哪个目录。
如果是其他问题也请麻烦指出。

### What did you do before

运行环境为ubuntu 20.04
首先我根据文档中的前端开发环境的内容构建了环境即
```
git clone https://github.com/OpenXiangShan/xs-env
cd xs-env
sudo -s ./setup-tools.sh 
source setup.sh 
source update-submodule.sh
```
之后我根据文档FPGA Prototype部分的编译香山昆明湖RTL对代码进行编译即
```
cd XiangShan
export NOOP_HOME=$(pwd)
git checkout 4b9ddb8a6311bff6f5e29cb3722d1f4236d66292
make init
make clean
make verilog CONFIG=FpgaDefaultConfig L2_CACHE_SIZE=256 L3_CACHE_SIZE=768
```
最后我根据后续步骤准备去生成 Vivado 项目，编译二进制流时发生了这个错误

### Environment

- XiangShan branch: master
- XiangShan commit id: 4b9ddb8a6311bff6f5e29cb3722d1f4236d66292
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: Mill Build Tool version 0.12.3
- java version: openjdk 11.0.26 2025-01-21

1024493603, 2025-04-09:
我在`env-scripts/xs_nanhu_fpga/src/rtl`下找到了这些文件，请问是否兼容昆明湖

[TRANSLATION]
I found these files under ` env scripts/xs_nanhu_fpga/src/rtl `. May I ask if they are compatible with Kunming Lake

xianfeili, 2025-04-10:
will fixed when fix_kmh_file_miss branch merge into the main

1024493603, 2025-04-10:
> will fixed when fix_kmh_file_miss branch merge into the main

感谢回复。
我更换到南湖分支进行编译之后遇到了一些问题。
首先我将mill更换为0.9.8版本编译verilog代码，之后创建vivado项目时 -part参数为`xcvu19p-fsva3824-2-e`，因为我使用FPGA的是Xilinx ZCU104，所以我将`env-scripts/xs_nanhu_fpga/src/rtl`内的文件的该参数改为了`xczu7ev-ffvc1156-2-i`。之后由于Vivado报脚本版本不匹配的问题，所以我将所有tcl文件中`set scripts_vivado_version`参数改为了我现在使用的2022.2。
最后创建项目时会报如下错误。

[TRANSLATION]
Thank you for your reply.
I encountered some issues after switching to the Nanhu branch for compilation.
Firstly, I replaced the mill with version 0.9.8 to compile the Verilog code. Then, when creating the Vivado project, the - part parameter was set to `xcvu19p-fsva3824-2-e`. Since I am using Xilinx ZCU104 FPGA, I changed the parameter of the file in ` env-scripts/xs_nanhu_fpga/src/rtl ` to ` xczu7ev-ffvc1156-2-i `. Later, due to the issue of script version mismatch reported by Vivado, I changed the `set scripts_vivado_version` parameter in all TCL files to the one I am currently using in 2022.2.
The following error will be reported when creating the project.

[image]

请问这是由于我某个步骤导致的错误吗？

[TRANSLATION]
Is this an error caused by one of my steps？

xianfeili, 2025-04-11:
[image]

You can comment this line in the tcl. The xdma block design is not use in the minimal system. Maybe it can pass the project generate.

It's prefer to use vivado 2020.2 to generate the project and upgrade it to your own vivado version.

1024493603, 2025-04-17:
我按照您的方式注释了这一行，然后成功构建了vivado项目，但是在生成bitstream时仅运行了很短的时间就结束了。
我按照 https://github.com/OpenXiangShan/XiangShan/issues/3782 检查了我的环境及项目。
`$NOOP_HOME`环境是正确的，vivado也没有在后台执行，没有生成`xs_nanhu/xs_nanhu.runs/impl_1/runme.log`文件，执行`make bitstream CORE_DIR=$NOOP_HOME/build`的输出信息如下。没有成功生成脚本中的bit 及 ltx 文件。

[TRANSLATION]
I annotated this line in your way and successfully built the Vivado project, but it only ran for a short time when generating the bitstream and ended.
I checked my environment and project using https://github.com/OpenXiangShan/XiangShan/issues/3782 .
The `$NOOP-HOME` environment is correct, and Vivado did not execute it in the background. The file `xs_nanhu/xs_nanhu.runs/impl_1/runme.log` was not generated. The output information of executing `make bitstream CORE_DIR=$NOOP_HOME/build` is as follows. The bit and ltx files in the script were not successfully generated.
[log.txt](https://github.com/user-attachments/files/19791602/log.txt)

vivado的GUI打开项目之后显示如下

[TRANSLATION]
After opening the project on Vivado's GUI, it displays as follows
[image]

xianfeili, 2025-04-17:
You can try source the blk_mem_gen_0.tcl manually. If it faild too. It may be your device don't support some IP that 19P used.