OpenXiangShan/XiangShan #4544: Load/Hybrid Unit, `full_fwd` signal redundant/misleading

plutsyk, 2025-04-11:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Given the definition of `s2_data_fwded` in [LoadUnit.scala#L1311](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/mem/pipeline/LoadUnit.scala#L1311), it seems that the `s2_out.rep_info.full_fwd` is driven constant zero. Unless I am missing something, this signal is used only in [LoadQueueReplay.scala#L695](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/mem/lsqueue/LoadQueueReplay.scala#L695), to reset the corresponding "blocking pointer". As it is used only when `s2_dcache_miss` is high (i.e. only when "full" data is not available through forwarding), perhaps there is no need for the `full_fwd` altogether...

This also concerns `s2_data_fwded` in [HybridUnit.scala#L966](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/mem/pipeline/HybridUnit.scala#L966), where the signal actually indicates presence of tag-errors (which no longer raise exceptions in `HybridUnit` it seems). In this case, perhaps the `full_fwd` should be renamed to something that reflects the meaning slightly better.

Anzooooo, 2025-04-14:
`Given the definition of s2_data_fwded in LoadUnit.scala#L1311, it seems that the s2_out.rep_info.full_fwd is driven constant zero. ` This is wrong.
As you say later, that signal is used for judgement in `LoadQueueReplay`.
When `dcache miss`, if my data is all from `fwd`, there is no need to block in `LoadQueueReplay`.

We think the name is still appropriate and, at the moment, we don't generally make changes to naming issues.
If you have a more appropriate naming, feel free to give it directly.

plutsyk, 2025-04-14:
Thank you for the reply.

Could you please clarify the intended behavior a little more... namely, can the `LoadQueueReplay` still "see" a miss when the data was successfully forwarded? I think it cannot (in the current version [LoadUnit.scala#L1242](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/mem/pipeline/LoadUnit.scala#L1242) defines `val s2_dcache_miss = io.dcache.resp.bits.miss && !s2_fwd_frm_d_chan_or_mshr && !s2_full_fwd && !s2_in.nc`).

Anzooooo, 2025-04-15:
Yes, the effect here is that there is no need for `LoadQueueReplay` to see the `dcache miss` if it can be fully forward passed.
This logic is intuitive, because if we can get the data we need via a forward pass, we don't need to get it from the dcache anymore.

closed 2025-09-05 by Yan-Muzi