OpenXiangShan/XiangShan #4548: Mismatch in write `mip` between XiangShan and NEMU

LuLuji04, 2025-04-11:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

Here is a mismatch in the write behavior of the **`mip`** register between **XiangShan** and **NEMU**
Here is the disassembly of the instruction:

[image]

Here is the log of difftest:
[image]
[image]

### Expected behavior

- The value of `mip` should be correctly written as expected from the behavior of NEMU

### To Reproduce

Here is the source code, binary file:
[test.zip](https://github.com/user-attachments/files/19712522/test.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: cb62726bba95608b79405d17edb74f5ea430c4f2
- NEMU commit id: a6a5f9b1452adef3b90a0a47f59fd38a61df1e9b
- SPIKE commit id:6268f2ead5595a40af1e27943f42a39846a37c9d

NewPaulWalker, 2025-04-14:
Duplicate with https://github.com/OpenXiangShan/XiangShan/issues/4549, I was able to pass the test using the latest master. So please try testing again with the latest versions of XiangShan and NEMU.

LuLuji04, 2025-04-14:
Thank you for the clarification. I will retest using the latest versions of XiangShan and NEMU. Now this issue can be closed.

closed 2025-04-14 by LuLuji04