OpenXiangShan/XiangShan #4549: Mismatch in `medeleg` bit 7 between XiangShan and NEMU

LuLuji04, 2025-04-11:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

When setting bit 7 of the **`medeleg`** register, there is a mismatch in the value of **bit 7** between **XiangShan** and **NEMU** 
Here is the disassembly of the instruction:
[image]
Here is the log of difftest:
[image]
[image]

### Expected behavior

- The value of **bit 7** in the **`medeleg`** register should be consistent across both XiangShan and NEMU

### To Reproduce

Here is the source code, binary file:
[test.zip](https://github.com/user-attachments/files/19712751/test.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: cb62726bba95608b79405d17edb74f5ea430c4f2
- NEMU commit id: a6a5f9b1452adef3b90a0a47f59fd38a61df1e9b
- SPIKE commit id:6268f2ead5595a40af1e27943f42a39846a37c9d

NewPaulWalker, 2025-04-14:
Thanks for your issue.

I noticed that the XiangShan commit you used is dated Oct 30, 2024, while the NEMU commit is from Aug 25, 2024. These two may not be compatible. I tested your test case using the latest master branches — 
XiangShan commit: 9feb8e87b9aadf4986597036dcf36d34a2762141
NEMU commit: e5b5811ab1a3f1e696a2976e8ba04deb8d4eb2c4
and the test passed successfully. 

**So please try testing again with the latest versions of XiangShan and NEMU.**

Additionally, we usually refer to bit 0 as the least significant bit, so the discrepancy here should be described as Bit 3 instead of Bit 4.

NewPaulWalker, 2025-04-14:
> Thank you for the clarification. I apologize for not updating to the latest commit when testing earlier. I will proceed with testing using the latest versions of XiangShan and NEMU. Additionally, it might be helpful to add a note in the issue template to remind everyone to use the latest commit.

Thanks for your suggestion! We've prepared to add a reminder to update the code in the issue template.

closed 2025-04-15 by Tang-Haojin