OpenXiangShan/XiangShan #4568: Segmentation fault when running simple example (hello) on emu

rseac, 2025-04-15:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reproduced the problem using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了问题。

### Describe you problem

I'm trying to run an application on the core.  When running `emu` with the nexus-am example of hello (or any other for that matter), I get a segmentation fault.

```
./emu -i $AM_HOME/apps/hello/build/hello-riscv64-xs.bin
emu compiled at Apr 12 2025, 18:00:30
Using simulated 32768B flash
Using simulated 8386560MB RAM
The image is /xs-env/nexus-am/apps/hello/build/hello-riscv64-xs.bin
The reference model is /xs-env/NEMU/build/riscv64-nemu-interpreter-so
Segmentation fault (core dumped)
```

### What did you do before

Compiled the application:

cd /xs-env/
source env.sh

cd NEMU && make clean && make riscv64-xs-ref_defconfig

cd /xs-env/nexus-am/apps/hello
make ARCH=riscv64-xs

cd /xs-env/XiangShan 
make emu CONFIG=MinimalConfig EMU_TRACE=1 -j32

cd /xs-env/XiangShan/build
./emu -i $AM_HOME/apps/hello/build/hello-riscv64-xs.bin
```

The output I get is below.
```
emu compiled at Apr 12 2025, 18:00:30
Using simulated 32768B flash
Using simulated 8386560MB RAM
The image is /xs-env/nexus-am/apps/hello/build/hello-riscv64-xs.bin
The reference model is /xs-env/NEMU/build/riscv64-nemu-interpreter-so
Segmentation fault (core dumped)
```

This happens with all benchmarks I try to run ...

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

NewPaulWalker, 2025-04-16:
Thanks for your issue!

It seems you only ran the command `make riscv64-xs-ref_defconfig` when compiling NEMU. In fact, you also need to run `make` or `make -j` afterwards. Please make sure that the file `/xs-env/NEMU/build/riscv64-nemu-interpreter-so` exists — that might help resolve the problem.

rseac, 2025-04-16:
I had mainly forgotten to write that I did make -j.  I had done it, and the segmentation fault still persists.

Notice in the log I printed `The reference model is /xs-env/NEMU/build/riscv64-nemu-interpreter-so`.  So, that does exist.

NewPaulWalker, 2025-04-16:
Sure, you should provide the environment you're using, including but not limited to the commit IDs of XiangShan, NEMU, and nexus-am. This is actually something required by our issue template as well.

> ### Environment
> 
> - XiangShan branch: 
> - XiangShan commit id: 
> - NEMU commit id: 
> - SPIKE commit id: 
> - Operating System: 
> - gcc version: 
> - mill version: 
> - java version:

rseac, 2025-04-16:
Absolutely, I went ahead and updated the original post with the information that I have.

NewPaulWalker, 2025-04-18:
Sorry, but based on the XiangShan commit and NEMU commit you're using, as well as the configs used by both XiangShan and NEMU, I still can't reproduce the segmentation fault issue.

You might consider uploading the binary file(hello-riscv64-xs.bin) and the complete log so we can take a closer look.

In the meantime, you could also try running other binaries from the `XiangShan/ready-to-run` directory with `emu` to see if the same error occurs.

NewPaulWalker, 2025-04-23:
> When I try to run `XiangShan/ready-to-run` I still get the segmentation fault.

Alright, then perhaps we can consider that this might not be a problem with the workload...

> Just a note, I'm building this within a docker image; not sure if that causes some odd behaviour.

I'm not sure if it's related, because this issue is just too strange.

> How do I dump out the entire log? When I provide the `--log-begin` flag, it still doesn't print anything.

What I mean is the runtime log is basically just the command-line output you provided... doesn't seem to have any additional information.

I currently have two ideas that you might need to try out yourself(since I really can't reproduce this issue):

1. You can try running this command
`./emu -i $NOOP_HOME/ready-to-run/microbench.bin --no-diff`
it will make the emulator `emu` skip differential testing, which means it won't require `riscv64-nemu-interpreter-so`. This might help determine whether the issue lies in XiangShan, Difftest, or NEMU itself.

2. You can add debugging options when compiling the emulator `emu`, and then use GDB to debug it. After the segmentation fault occurs, you can inspect the stack frames to locate the root cause of the issue.

2.1 add debugging options `-g` in file `/xs-env/XiangShan/difftest/Makefile`

``` diff
diff --git a/Makefile b/Makefile
index f354440d..73b44552 100644
--- a/Makefile
+++ b/Makefile
@@ -239,6 +239,8 @@ ifeq ($(CXX_NO_WARNING),1)
 SIM_CXXFLAGS += -Werror
 endif
 
+SIM_CXXFLAGS += -g
+
 include verilator.mk
 include gsim.mk
 include vcs.mk
```

2.2 Recompile the emulator
``` bash
cd /xs-env/XiangShan
make clean
rm -rf out
make emu CONFIG=MinimalConfig EMU_TRACE=1 -j32
```

2.3 Use GDB to debug the emu program.
``` bash
cd /xs-env/XiangShan/build
gdb emu # Please make sure gdb can read symnols from emu...

#in gdb
set args -i ../ready-to-run/microbench.bin
r
bt # After segmentation fault occurs, use GDB to identify the root cause of the issue.
info frame
```

Also, I'd like to ask you to confirm whether you've run the `make init` command in the XiangShan directory—this will switch the submodules in the XiangShan directory to their corresponding versions.

In the XiangShan commit c01e75b55fbe706c21954429f2ec968cbc61a2cc that you are using, the corresponding submodule version should be:
``` bash
git submodule status                                 
 6608dc16ec944800d39200e2fad5924d0968b42b ChiselAIA (native_axi-36-g6608dc1)
 b2b3ae7d6a67c5d527e8933d138e616afe4a0ea2 coupledL2 (v0.9-254-gb2b3ae7)
 9d5e8eb361354d0be33c1eccff9f79fee67cca67 difftest (heads/master)
 e1bd4695ca7beb36a5ce7357e9527ad9e95b9ec1 fudian (heads/pipeline)
 c5e02bd61625fd53ad2410999ff7d57a61234230 huancun (heads/master)
 614ceb4c5d39d32e6ae786693a35da429ea32c12 openLLC (614ceb4)
 75d8aeb40e5302a1c2e7c517eb07e4e6f8a2efac ready-to-run (remotes/origin/fix-dbltrp2-43-g75d8aeb)
 bea0af5d855a871699b907c1988961141bbf28e7 rocket-chip (heads/master)
 baacebea4bf173b2b676e99f5bb0f9abf10a5f48 utility (heads/master)
 cadd3c2f43096e253d61296b33ac697be8354e29 yunsuan (remotes/origin/fix_pipeline-11-gcadd3c2)
```

rseac, 2025-04-25:
Thank you for the detailed debugging suggestions.

1. When running `./emu -i $NOOP_HOME/ready-to-run/microbench.bin --no-diff`, the benchmark runs, and does not segfault.
 
2. I did run `emu` through `gdb`.  Here is the output.
```
root@52b9f945e239:/xs-env/XiangShan/build# gdb ./emu
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.2) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./emu...
(gdb) set args -i ../ready-to-run/microbench.bin
(gdb) r
Starting program: /xs-env/XiangShan/build/emu -i ../ready-to-run/microbench.bin
warning: Error disabling address space randomization: Function not implemented
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
emu compiled at Apr 24 2025, 20:42:56
[New Thread 0x155554ba5700 (LWP 19403)]
[New Thread 0x1555549a4700 (LWP 19404)]
[New Thread 0x1555547a3700 (LWP 19405)]
[New Thread 0x1555545a2700 (LWP 19406)]
[New Thread 0x1555543a1700 (LWP 19407)]
[New Thread 0x1555541a0700 (LWP 19408)]
[New Thread 0x155553f9f700 (LWP 19409)]
[New Thread 0x155553d9e700 (LWP 19410)]
[New Thread 0x155553b9d700 (LWP 19411)]
[New Thread 0x15555399c700 (LWP 19412)]
[New Thread 0x15555379b700 (LWP 19413)]
[New Thread 0x15555359a700 (LWP 19414)]
[New Thread 0x155553399700 (LWP 19415)]
[New Thread 0x155553198700 (LWP 19416)]
[New Thread 0x155552f97700 (LWP 19417)]
[New Thread 0x155552d96700 (LWP 19418)]
[New Thread 0x155552b95700 (LWP 19419)]
[New Thread 0x155552994700 (LWP 19420)]
[New Thread 0x155552793700 (LWP 19421)]
[New Thread 0x155552592700 (LWP 19422)]
[New Thread 0x155552391700 (LWP 19423)]
[New Thread 0x155552190700 (LWP 19424)]
[New Thread 0x155551f8f700 (LWP 19425)]
[New Thread 0x155551d8e700 (LWP 19426)]
[New Thread 0x155551b8d700 (LWP 19427)]
[New Thread 0x15555198c700 (LWP 19428)]
[New Thread 0x15555178b700 (LWP 19429)]
[New Thread 0x15555158a700 (LWP 19430)]
[New Thread 0x155551389700 (LWP 19431)]
[New Thread 0x155551188700 (LWP 19432)]
[New Thread 0x155550f87700 (LWP 19433)]
[New Thread 0x155550d86700 (LWP 19434)]
[New Thread 0x155550b85700 (LWP 19435)]
[New Thread 0x155550984700 (LWP 19436)]
[New Thread 0x155550783700 (LWP 19437)]
[New Thread 0x155550582700 (LWP 19438)]
[New Thread 0x155550381700 (LWP 19439)]
[New Thread 0x155550180700 (LWP 19440)]
[New Thread 0x15554ff7f700 (LWP 19441)]
[New Thread 0x15554fd7e700 (LWP 19442)]
[New Thread 0x15554fb7d700 (LWP 19443)]
[New Thread 0x15554f97c700 (LWP 19444)]
[New Thread 0x15554f77b700 (LWP 19445)]
[New Thread 0x15554f57a700 (LWP 19446)]
[New Thread 0x15554f379700 (LWP 19447)]
[New Thread 0x15554f178700 (LWP 19448)]
[New Thread 0x15554ef77700 (LWP 19449)]
Using simulated 32768B flash
Using simulated 8386560MB RAM
The image is ../ready-to-run/microbench.bin
The reference model is /xs-env/NEMU/build/riscv64-nemu-interpreter-so

Thread 1 "emu" received signal SIGSEGV, Segmentation fault.
0x00000555ff484204 in allocate_memory_with_mmap () from /xs-env/NEMU/build/riscv64-nemu-interpreter-so
(gdb) bt
#0  0x00000555ff484204 in allocate_memory_with_mmap () from /xs-env/NEMU/build/riscv64-nemu-interpreter-so
#1  0x00000555ff48424f in init_mem () from /xs-env/NEMU/build/riscv64-nemu-interpreter-so
#2  0x00000555ff47530f in difftest_init () from /xs-env/NEMU/build/riscv64-nemu-interpreter-so
#3  0x000000000041f1e8 in RefProxy::RefProxy (this=0xd008f60, coreid=-2147483648, ram_size=3, env=0x155554eccb06 <__GI___mmap64+38> "H=",
    file_path=0xffffffff <error: Cannot access memory at address 0xffffffff>) at /xs-env/XiangShan/difftest/src/test/csrc/difftest/refproxy.h:186
#4  NemuProxy::NemuProxy (this=0xd008f60, coreid=-2147483648, ram_size=3) at /xs-env/XiangShan/difftest/src/test/csrc/difftest/refproxy.cpp:319
#5  0x0000000000417b5c in Difftest::update_nemuproxy (this=0xd008b00, coreid=0, ram_size=0) at /xs-env/XiangShan/difftest/src/test/csrc/difftest/difftest.cpp:207
#6  init_nemuproxy (ramsize=3) at /xs-env/XiangShan/difftest/src/test/csrc/difftest/difftest.cpp:57
#7  0x000000000041f8d1 in Emulator::Emulator (this=0xcffb2c0, argc=<optimized out>, argv=<optimized out>) at /xs-env/XiangShan/difftest/src/test/csrc/verilator/emu.cpp:481
#8  0x0000000000413ebb in main (argc=-2147483648, argv=0x7fffffffe5b8) at /xs-env/XiangShan/difftest/src/test/csrc/common/main.cpp:40
```

3. Yes, I am running make init.

Tang-Haojin, 2025-04-25:
Could you provide the value of `/proc/sys/vm/overcommit_memory` and your size of RAM? Besides you can try to share your `emu` and `nemu-so` with us.

rseac, 2025-04-28:
Note that this is within a docker container.

1. overcommit_memory
```
$ cat /proc/sys/vm/overcommit_memory 
0
```

2. Size of ram
```
cat /proc/meminfo | grep MemTotal
MemTotal:       263642608 kB
```

I'll upload the emu and nemu-so files soon.

rseac, 2025-05-02:
Thank you @Tang-Haojin and @NewPaulWalker  for your continued support on this.  Although I wasn't able to really figure out what was happening, I was able to build the example and run it successfully on a different system.  Hence, I'm going to close this issue.  Thank you again!

closed 2025-05-02 by rseac

chenhychen, 2025-08-27:
I have the same problem using VCS with either reference models built from scratch or from `ready-to-run`. Could you please help?

### Environment

- XiangShan branch: master
- XiangShan commit id: d85795349c7a0f8ef4efa272f5dfafd549f52d53
- NEMU commit id: 53bcb5686f8fd05248ae98546b7dc04bdca1bbb0
- SPIKE commit id: 
- Operating System: NAME="Red Hat Enterprise Linux"
                                 VERSION="8.10 (Ootpa)"
                                 ID="rhel"
                                 ID_LIKE="fedora"
                                 VERSION_ID="8.10"
- gcc version: 14.2.0
- mill version: 0.12.3
- java version: 21.0.7
- vcs version: 2023.12

### git submodule status
1d2a9bf6da8e050d975ef2669c42faa93943f489 DRAMsim3 (heads/cosim-kmh)
53bcb5686f8fd05248ae98546b7dc04bdca1bbb0 NEMU (remotes/origin/wzz-debug-tmp-cc72-79-g53bcb568)
fc12171d929e7e589fab9f794ab63ce12e6c594e NutShell (release-211228-126-gfc12171)
16ae9ddcda54fc9a2fddffad73174cf793ac7814 XiangShan (v3.2.2-alpha-587-g16ae9ddcd)
086a238062fdfb3281b745a537b361983d0154b5 nexus-am (remotes/origin/maprobe-230305-43-g086a2380

cat /proc/sys/vm/overcommit_memory: 0
size of RAM: 1TB

### Simulation log
```
./build/simv +workload=xs-env/nexus-am/apps/hello/build/hello-riscv64-xs.bin +dump-wave=vpd
Info: [VCS_SAVE_RESTORE_INFO] ASLR (Address Space Layout Randomization) is detected on the machine. To enable $save functionality, ASLR will be switched off and simv re-executed.
Please use '-no_save' simv switch to avoid re-execution or '-suppress=ASLR_DETECTED_INFO' to suppress this message.
Chronologic VCS simulator copyright 1991-2023
ram image: xs-env/nexus-am/apps/hello/build/hello-riscv64-xs.bin
Using simulated 8386560MB RAM
The image is xs-env/nexus-am/apps/hello/build/hello-riscv64-xs.bin
Using simulated 32768B flash
The reference model is xs-env/NEMU/build/riscv64-nemu-interpreter-so

An unexpected termination has occurred in ./build/simv due to a signal: Segmentation fault

Command line: ./build/simv +workload=xs-env/nexus-am/apps/hello/build/hello-riscv64-xs.bin +dump-wave=vpd

--- Stack trace follows:

Dumping VCS Annotated Stack:
#0  0x000015551d8f4eeb in waitpid () from /lib64/libc.so.6
#1  0x000015551d848747 in do_system () from /lib64/libc.so.6
#2  0x0000155526b7ea80 in SNPSle_10ee25eff68cd8461c9146fa1d0b35e87067f3c8015b313e639d2928478c79b3f673f99203bcf8be64600612100082236bffb2007f1e0ef9 () from vcs/V-2023.12-SP2-2/linux64/lib/liberrorinf.so
#3  0x0000155526b805c4 in SNPSle_10ee25eff68cd8461c9146fa1d0b35e87067f3c8015b313efba706aab251478fa49e66610e453774633a6c152e7ef778f93045171fb3645d () from vcs/V-2023.12-SP2-2/linux64/lib/liberrorinf.so
#4  0x0000155526b78653 in SNPSle_d35ca1ff70d465c2b9b1a72eee90a50630165806651fae96c1bbda5b5d02066c () from vcs/V-2023.12-SP2-2/linux64/lib/liberrorinf.so
#5  0x0000155521909cfb in SNPSle_64133461705005bb725549e2e6fa1b3f () from vcs/V-2023.12-SP2-2/linux64/lib/libvcsnew.so
#6  0x000015552134d14d in SNPSle_82244d58c54c18c70d63edc9becab634 () from vcs/V-2023.12-SP2-2/linux64/lib/libvcsnew.so
#8  0x000005559ad07020 in allocate_memory_with_mmap () from xs-env/NEMU/build/riscv64-nemu-interpreter-so
#9  0x000005559ad0706b in init_mem () from xs-env/NEMU/build/riscv64-nemu-interpreter-so
#10 0x000005559ad049db in difftest_init () from xs-env/NEMU/build/riscv64-nemu-interpreter-so
#11 0x000000000042f23e in AbstractRefProxy::AbstractRefProxy(int, unsigned long, char const*, char const*) ()
#12 0x000000000042f96f in NemuProxy::NemuProxy(int, unsigned long) ()
#13 0x000000000042c2f6 in Difftest::update_nemuproxy(int, unsigned long) ()
#14 0x000000000042c336 in init_nemuproxy(unsigned long) ()
#15 0x000000000043da13 in simv_init ()
#16 0x000015552a64249f in F_VCSgd_ygb0U_19_0 () from xs-env/XiangShan/build/simv.daidir/_3612837_archive_1.so
#17 0x000015552a64222f in F_VCSgd_ygb0U_22_0 () from xs-env/XiangShan/build/simv.daidir/_3612837_archive_1.so
#18 0x0000155522b6115f in SNPSle_64f77020d05c8e4ca0d9aedaf17169817b52ebd39798ba2c () from vcs/V-2023.12-SP2-2/linux64/lib/libvcsnew.so
#19 0x00001555215c1848 in SNPSle_63c7abb936d47b07d29d61775f9dbe8c1486beb9213552a3 () from vcs/V-2023.12-SP2-2/linux64/lib/libvcsnew.so
#20 0x00001555219235ba in SNPSle_490598bfebcc8e81efcf3d1815066cc9 () from vcs/V-2023.12-SP2-2/linux64/lib/libvcsnew.so
#21 0x0000155521941945 in SNPSle_490598bfebcc8e81 () from vcs/V-2023.12-SP2-2/linux64/lib/libvcsnew.so
#22 0x000000000043dfdf in main ()

Process VmPeak: 22902311820 kb, VmSize: 22902311784 kb
System Free Memory: 483995180 kb, System Free Swap: 30273432 kb

No context available
```

I am unsure of the reasons, but I am unable to add my Nemu-so file, which I built from scratch.

Thank you very much!