OpenXiangShan/XiangShan #4575: `amomin.w` Instruction Behavior Inconsistency

LuLuji04, 2025-04-16:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。
- [x] I have reproduced the incorrect behaviors using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了不正确的行为。

### Describe the bug

At the given program counter (`pc = 0x00800004ee`), the value of `a3` differs between the correct and incorrect states. The expected value of `a3` is `0x0000000000000000`, while the actual value is `0x0000000000800000`.
[image]
[image]

### Expected behavior

The value in register `a3` should be `0x0000000000000000` after executing the `amomin.w` instruction.

### To Reproduce

Here is the source code, binary file:
[test.zip](https://github.com/user-attachments/files/19783487/test.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: c01e75b55fbe706c21954429f2ec968cbc61a2cc
- NEMU commit id: 738ae334edc62237c3d4b49c722882ee1b7324fc

Anzooooo, 2025-04-18:
Thanks for bringing it up, it was a nemu bug that I've fixed here (https://github.com/OpenXiangShan/NEMU/pull/854)

LuLuji04, 2025-04-18:
Thank you very much for your confirmation. This indeed seems to be a bug in `NEMU`. As I used the `ready-to-run/riscv64-spike-so `as a reference, it fell into an infinite loop without triggering the mismatch. I would greatly appreciate it if you could explain how you pinpointed it [OpenXiangShan/NEMU#854](https://github.com/OpenXiangShan/NEMU/pull/854)?

NewPaulWalker, 2025-04-21:
After discovering an inconsistency between XiangShan and NEMU, I conducted differential testing between XiangShan and Spike. However, it seemed to fall into an infinite loop (though it successfully passed the point where it previously disagreed with NEMU), which strongly suggests that the issue is a bug in NEMU.

Then I ran differential testing between NEMU and Spike and found that the amomin.w instruction was just where the diff was reported; the actual root cause was a preceding amocas.d instruction. I noticed that it fetched a non-zero operand from the x0 register, which caused the instruction to skip writing a value to memory. This eventually led to the failure of the subsequent amomin instruction. So the issue appears to be a decoding bug.

Additionally, it doesn't seem like the amocas.d instruction was deliberately generated. From what I observed, after an ebreak instruction triggered an exception, you simply incremented mepc by 4 and did an mret, which caused the PC to land in the middle of an instruction.

closed 2025-04-21 by Anzooooo