OpenXiangShan/XiangShan #4576: `unimp` After Returning from `ebreak` and Continuing Instructions Trigger Mismatch

LuLuji04, 2025-04-16:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。
- [x] I have reproduced the incorrect behaviors using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了不正确的行为。

### Describe the bug

After returning from the `ebreak` instruction, XiangShan fails to correctly recognize the next instruction, misidentifying it as `c.unimp` instead of the expected instruction. The misidentification of the instruction as `c.unimp` triggers an illegal instruction exception. Following this, XiangShan continues to execute instructions, and when an exception is manually triggered (such as `lw a0, 0(s3)`), it leads to a mismatch in the CSR registers.
[image]
[image]

### Expected behavior

After returning from `ebreak`, XiangShan should correctly recognize the next instruction.

If the behavior of recognizing `unimp` is correct, the remaining instruction operations should also be consistent with NEMU.

### To Reproduce

Here is the source code, binary file:
[test.zip](https://github.com/user-attachments/files/19783759/test.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: c01e75b55fbe706c21954429f2ec968cbc61a2cc
- NEMU commit id: 738ae334edc62237c3d4b49c722882ee1b7324fc

Anzooooo, 2025-04-18:
Thanks for bringing it up, it was a nemu bug that I've fixed here (https://github.com/OpenXiangShan/NEMU/pull/854)

LuLuji04, 2025-04-18:
I believe Spike also has this issue, as I used the ready-to-run/riscv64-spike-so as a reference, and it also triggered this mismatch. It seems that there's a significant issue with NEMU‘s and Spike's exception handling, as other manually triggered exceptions are also being misidentified. For instance, both Spike and NEMU treat `ecall` as an illegal operand.

[image]
[image]

LuLuji04, 2025-04-18:
Additionally, it seems that after the mret return, the PC is stuck in the middle of an instruction, triggering an unimp exception. This implementation in XiangShan is correct. While from a software perspective, how should the exception handler determine how much to add to mepc to ensure it jumps to the correct address after the mret return?

NewPaulWalker, 2025-04-21:
> Thank you very much for your confirmation! It indeed appears to be a bug in `Xiangshan`, as I used Spike as a reference and observed the same mismatch. You can refer to this Spike issue for an explanation of the cause: [Spike Issue #1964](https://github.com/riscv-software-src/riscv-isa-sim/issues/1964)
> 
> I guess this issue is caused by `Cache Consistency Problem`. I enabled the instruction trace in `NEMU`, and you can observe the inconsistency in instruction fetching between the two. <img alt="Image" width="416" src="https://private-user-images.githubusercontent.com/115395701/435353397-cf58fe3d-7afa-4b1b-a6e6-9fe555cf2bc4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDUyMDc2MzUsIm5iZiI6MTc0NTIwNzMzNSwicGF0aCI6Ii8xMTUzOTU3MDEvNDM1MzUzMzk3LWNmNThmZTNkLTdhZmEtNGIxYi1hNmU2LTlmZTU1NWNmMmJjNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNDIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDQyMVQwMzQ4NTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNTBmYjNjMTBlYWQyZjYxMzI2NWQzOTNhNWRmMjI3MzVhNmVmOTdmNzMzMzY4NzZmOWVmMzZhYTUzYWVhYmU2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.BZs0PQO2ia874AJZqcCWMemRMbkwPZjH8KBkQazYO-I"> <img alt="Image" width="422" src="https://private-user-images.githubusercontent.com/115395701/435353398-831be2cb-22e6-4ef8-861e-a12b9bcc2e56.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDUyMDc2MzUsIm5iZiI6MTc0NTIwNzMzNSwicGF0aCI6Ii8xMTUzOTU3MDEvNDM1MzUzMzk4LTgzMWJlMmNiLTIyZTYtNGVmOC04NjFlLWExMmI5YmNjMmU1Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNDIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDQyMVQwMzQ4NTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05ZTMyNDZkOTA2ZTM0YWZmYzgzNWE1MGM4NGM4NzkzY2M3YTljYTIwYjRhMzZiZTc0NzQ5NzE1NTliZjRjY2Y4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.KkrxRncxqqoruVZFx5xYfd8EMo2wHgNmdhsu4AFylQ8">

Yes, this is indeed a self-modifying program. For programs like this, it's true that XiangShan, NEMU, and Spike may exhibit inconsistent behavior. The reason is that XiangShan writes store data into the DCache, but it fetches instructions from the ICache, which may still hold the old value. On the other hand, NEMU doesn't have a cache structure like this, so it fetches the updated instruction directly, leading to the behavioral mismatch.

For self-modifying programs, you should insert a `fence` instruction afterward to synchronize memory.

If you filter the issues in the XiangShan repository using the label `not bug`, you'll find many issues related to self-modifying code...

NewPaulWalker, 2025-04-21:
> Additionally, it seems that after the `mret` return, the PC is stuck in the middle of an instruction, triggering an `unimp` exception. This implementation in XiangShan, as well as NEMU and Spike, is consistent. While, from a software perspective, how should the exception handler determine how much to add to `mepc` to ensure it jumps to the correct address after the `mret` return?

If your handler for illegal instructions (exception code 2) simply aims to skip the current instruction, you can use the value in `mtval` (which holds the encoding of the illegal instruction) to determine whether it is a C extension (compressed) instruction, and based on that, decide whether to increment `mepc` by 4 or 2.

You can refer to the RISC-V manual to understand the encoding patterns of C extension (compressed) instructions. Alternatively, you can choose not to compile code with C extension instructions or enforce 4-byte alignment for all instructions.

LuLuji04, 2025-04-25:
Thank you for your reply! Is there any way to align `NEMU` and `XiangShan` for self-modifying programs?

NewPaulWalker, 2025-04-25:
> Thank you for your reply! Is there any way to align `NEMU` and `XiangShan` for self-modifying programs?

Yes, yes, just as I mentioned earlier:

> For self-modifying programs, you should insert a `fence` instruction afterward to synchronize memory.

But you need to be careful: this fence instruction must not be self-modified.