OpenXiangShan/XiangShan #4589: the data.txt may has invalid RISCV instr

zourenDevote, 2025-04-18:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reproduced the problem using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了问题。

### Describe you problem

I have according to the document `integration/FPGA Prototype/Overview` and download the data.txt(version nanhu), i use a convert.cpp file to convert the data.txt to data.bin and use `ricv64-unknown-elf-objdump` to generate the asm code, and i found there have a invalid instr? i don't know why.

 

### What did you do before

the code `convert.cpp` is
`
#include <iostream>
#include <fstream>
#include <string>

using namespace std;

// 将单个十六进制字符转换为数值
unsigned char charsToHexByte(char c1, char c2) {
    // 将字符转换成对应的 16 进制数值
    unsigned char high = (c1 >= 'a') ? (c1 - 'a' + 10) : (c1 - '0');
    unsigned char low  = (c2 >= 'a') ? (c2 - 'a' + 10) : (c2 - '0');
    
    // 合并成一个 byte
    return (high << 4) | low;
}

void processLongHexString(const std::string& hexStr, std::ofstream  &binFile) 
{
    size_t len = hexStr.length();
    unsigned char byte = 0;
    for(int i = 0 ; i < hexStr.size() ; ++i) {
         byte = charsToHexByte(hextStr(i), hexStr(i+1));
         binFile.write(reinterpret_cast<const char *>(&byte), 1);
    }
}

int main()
{
    const char *data_file = "data.txt";
    const char *out_file = "data.bin";

    ifstream in{data_file};
    ofstream out{out_file, ios::binary};

    string addr, value;

    while (!in.eof())
    {
        getline(in, addr);
        getline(in, value);

        if (value.empty()) continue;

        processLongHexString(value, out);
     }

    in.close();
    out.close();
    
    return 0;
}

the compile commad is(i am sure the data.txt and convert.cpp in the same driectory)
`
g++ convert.cpp -o convert
./convert
../riscv64-unknown-elf-gcc/bin/riscv64-unknown-elf-objdump -D -b binary -m riscv --insn-width=4 data.bin --adjust-vma=0x80000000 > data.asm 
`

the data.asm's code is, and the 0x800000a8 is a invalid instr? 

`
0000000080000000 <.data>:
    80000000:	9755      	srai	a4,a4,0x35
    80000002:	a05d      	j	0x800000a8
   .......
    800000a6:	7300      	flw	fs0,32(a4)
    800000a8:	6f20200f 	0x6f20200f
    800000ac:	c600      	sw	s0,8(a2)
`

`

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Ma-YX, 2025-04-18:
Thanks for your issue.

As for your problem, there might be something wrong unexpectedly during converting txt file to bin file with your own `convert.cpp`.

We have the [`Docker Instruction`](https://docs.xiangshan.cc/zh-cn/latest/integration/docker/). We advise you to use the stable docker process provided instead of generating bin file by yourself.

zourenDevote, 2025-04-22:
Thanks for you anwser ! i have read the bin2fpgadata.cpp code, and my convert.cpp's logic has error

closed 2025-09-05 by Yan-Muzi