OpenXiangShan/XiangShan #4607: Developing workloads on bare metal devices

Liujiaqi-jlu, 2025-04-22:
### Before start

- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

Hello, I am developing workloads on bare metal machines in the Nanhu version that do not use an operating system, but I am unable to directly manage programs using processes and threads. May I ask if there is any good solution for this?

Tang-Haojin, 2025-04-22:
What do you mean by "manage programs"? If you want to do processes/threads scheduling and context switching, an operating system, although may be very simple like RT-Thread, is what you need.

Liujiaqi-jlu, 2025-04-23:
> What do you mean by "manage programs"? If you want to do processes/threads scheduling and context switching, an operating system, although may be very simple like RT-Thread, is what you need.

Hello, it's like this. I want to directly develop workloads on the FPGA version of Nanhu for some network communication work, which will involve managing memory buffers and writing network protocol stacks. I don't want to use an operating system, but without its support, I won't be able to develop workloads effectively using processes and threads. So I would like to ask if there is any good solution without the support of an operating system. Thank you!

Tang-Haojin, 2025-04-23:
What you want is really like what an OS is good at. We also have a tool named [gos](https://github.com/OpenXiangShan/gos). You may file an issue under this repo but I think this tool is actually already an OS.

Liujiaqi-jlu, 2025-04-26:
> What you want is really like what an OS is good at. We also have a tool named [gos](https://github.com/OpenXiangShan/gos). You may file an issue under this repo but I think this tool is actually already an OS.

Thank you for your response. I am currently trying to develop a workload based on a bare metal single threaded environment, and if necessary, I will consider adding an operating system in the future. I have another question, what are the library functions that can be directly used when developing workloads in C language in a bare metal environment, and which library functions cannot be directly used? thank you!