OpenXiangShan/XiangShan #4609: The nanhu FPGA Prototype phenomenon of FPGA on the board？

zourenDevote, 2025-04-22:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [x] I have reproduced the problem using the latest commit on the master branch. 我已经使用 master 分支最新的 commit 复现了问题。

### Describe you problem

our fpga prototye in nanhu 2core, branh is nanhu_clockdiv2, commit id is: [09d2a32142c64fca898e17c0b943e61ddc331958](https://github.com/OpenXiangShan/XiangShan/commit/09d2a32142c64fca898e17c0b943e61ddc331958) we found some problem (we changed the DDR type and clock, use vivado ip to generate clock)

### What did you do before

Hello, I verify the Nanhu 2 core code of the relevant branch on the board according to the FPGA prototype/Overview of the official Xiangshan document, I would like to ask which Linux version of the data.txt you provide (based on Nanhu)? What happens to UART if it is successful? What we have observed now is that AXI_Bridge waddr of this IP has always been 0, and the raddr of DDR has been stuck in 0x80000040, we have changed the model and bit width of DDR and used the IP inside VIVADO to generate the clock, if it is smooth, will the serial port have output?

At first, we suspected that the data in the ROM was not being read, but by grabbing the raddr in the AXI_Bridge through VIO, we can see that raddr goes from 0x10000000 to 0x10000080, and compares the initial value inside the ROM (using RISC-V to assemble and observe that rdata is indeed 0x80, a command to jump to 0x80000000, and the output of rdata is as expected). 

And observing that the raddr of DDR has indeed changed from 0x00000000 to 0x80000000 and then has remained unchanged until 0x80000040, is this phenomenon normal? If not, do you have any suggestions?

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

xianfeili, 2025-04-22:
Can you provide the output data that read from DDR ? You can compare the data with the data.txt second line. Check the data that write into the DDR is right.

The following attachment is the boot log we capture on our own env. You can use to check the Linux version 

[nhv2_boot.log](https://github.com/user-attachments/files/19845357/nhv2_boot.log)

zourenDevote, 2025-04-22:
Thanks for your answer! We use vio check the signal of ip 'jtag_ddr_wrapper...' SOC_M_AXI_rdata and we have compared the SOC_M_AXI_rdata to data.txt, the data is right, but the SOC_M_AXI_raddr not changed in 0x80000040。

zourenDevote, 2025-04-22:
my environment is offline, could you provide the linux.bin to me, so i can assembly the linux.bin to look the riscv instr

zourenDevote, 2025-04-22:
the nhv2_boot.log file is the output of the data.txt? the addressing mapping of UART0 is what, i see the address is 0x310b0000 in the xiangshan document, `clint addr` means what, check the ddr, xmda, peripherals's base address?

zourenDevote, 2025-04-22:
sorry i look the nhv2_boot.log carefully and have found the output `clint addr: 0x310b0000`, and `clint addr: 0x80000000`, but another addr means what? i don't know

zourenDevote, 2025-04-22:
i used tcl script is `env_script/fpga_nanhu/tcl/xs_uart.tcl` and change the property of `jtag_ddr_subsys.tcl`

xianfeili, 2025-04-23:
https://docs.xiangshan.cc/zh-cn/latest/integration/fpga/fpga/#xiangshan-cfg-axi
Here is the memmap of Xiangshan SOC.
https://docs.xiangshan.cc/zh-cn/latest/integration/docker/
This document can help you generate the bin file.

closed 2025-09-05 by Yan-Muzi